microsoft/AirSim #181: where to find recorded images and pose?

gwnudt, 2017-04-26:
I follow the first way to gather the data for training some algorithm by simply pressing the record button on the lower right corner. But I can not find the recored images and poses for training some machine learning algorithms. Where are the  data stored and how we get the images and read poses?

saihv, 2017-04-27:
They should be stored in C:/Users/<username>/Documents/AirSim.

closed 2017-04-29 by sytelus

nirmita123, 2017-07-14:
[image]

The recorded images in my case are being stored as a txt file and it's and empty txt(0 bytes) file.
Can someone please suggest a solution?
Thanks a lot

saihv, 2017-07-14:
Those are not the images, that's the ground truth that's recorded by default along with the images: so the images are not being recorded at all. Did you press 3 in the game window so all the camera views are displayed?

nirmita123, 2017-07-17:
Thanks @saihv . And yes I pressed 3 for all the camera views but still no difference .

nirmita123, 2017-07-17:
Thank you so much @saihv . Now , I'm able to get the images in the same folder .

nirmita123, 2017-07-17:
Also can you please tell me what do these values in the txt file mean?
[image]

saihv, 2017-07-17:
That corresponds to lines 34 to 38 in [RecordingFile.cpp](https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/Recording/RecordingFile.cpp). Timestamp, position and orientation of the drone.

nirmita123, 2017-07-18:
Thanks @saihv .It was very helpful.

shristi945, 2017-07-24:
Hi @saihv   After pressing 3 and recording button in the game window the camera view is obtained but I could not get the large no of images recorded only one single image get recorded and the txt file is still of zero(0 bytes) size and that single image is not as clear as that in the game window camera view. 
Please help.

[image]

the game window view after pressing 3 is like this:

[image]

but the saved image I opened from here  ....\Documents\AirSim\ is like this:

[image]

shristi945, 2017-07-31:
The issue for not getting the number of images and just getting one image instead is resolved. The images got overwritten again and again and only the final image left as output of recording.This is done By changing the code of RecordingFile.cpp
(AirSim/Unreal/Plugins/AirSim/Source/Recording/RecordingFile.cpp) .
By changing the local declaration of imageSavedOk to global only which maintains its value true if the image is saved and hence returning true in if condition so as to get the screenshot saved to . . . log message. Also the txt file of ground truth values do not have 0 byte size.
But the image quality that is being saved is still the same. This problem is not resolved, Please Help.
Thanks in Advance.