microsoft/AirSim #183: Where to find the scene in your demo (city/street scene)

cca2016, 2017-04-27:
I have just downloaded the demo called LandscapeMountains, but found it not suitable for my work.
Where to find the scene in your demo (the city/street scene, denser than LandscapeMountains)
Can it be found in Epic Games? What is the name of it?

AlexandreBorowczyk, 2017-04-27:
[Here](https://www.unrealengine.com/marketplace/modular-neighborhood-pack) but prepare your credit card. However, you can always find [other](https://www.unrealengine.com/marketplace/profile/Epic%20Games) free stuff

closed 2017-04-29 by sytelus

cca2016, 2017-06-19:
@AlexandreBorowczyk I have downloaded it and added to my landscape map project. But how to use it ? As i click play, there is no drone in the scene and the scene is just like a demo movie. What should i do?

AlexandreBorowczyk, 2017-06-23:
did you follow the instruction of the demo?

lovettchris, 2017-06-23:
Specifically, if you skip to minute 4 in this [walkthrough video ](https://www.youtube.com/watch?v=HNWdYrtw3f0&feature=youtu.be) it shows you how to add AirSim to existing environment: