microsoft/AirSim #184: Can I simulate for my Hexrcopter of 1.6 meters?

liuli2008212, 2017-04-27:
I konw that airsim is support F450.But I want to simulate my Hexrcopter. Can I change the model?How to modify the model? 
Thank you.

lovettchris, 2017-04-27:
Sure, take a look at [Px4QuadX.hpp](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/vehicles/configs/Px4QuadX.hpp) and notice that it is calling initializeRotorQuadX from [MultiRotorParams.hpp](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/vehicles/MultiRotorParams.hpp), and notice that is just doing this 4 times:
````
    rotor_poses.emplace_back(VectorMath::rotateVector(Vector3r(-arm_lengths[3], 0, rotor_z), quadx_rot, true), 

      unit_z, RotorTurningDirection::RotorTurningDirectionCW);
````
So you could create your own hexacopter configuration which inherits from MultiRotorParams and adds 6 motors.  

So that takes care of the physics, then you'll also need to edit the AirSim Blueprint in the Unreal Editor to add a [3d mesh for Hexacopter](https://sketchfab.com/models/be5f7de9e6f546e5af72581696bbd4b4), which you may need to edit to make the rotors spin, and wire up the Unreal events to connect rotor angles to the mesh so they spin as per the physics...

This wiring of model to 3d view yappens in AFlyingPawn, notice it currently has a hard coded rotor_count of 4, you'll change that to 6.  Notice it is expecting the Unreal assets to contain a rotatable component named "Rotation0", "Rotation1"... as per the AFlyingPawn::setupComponentReferences method.

To edit the blueprint you'll need to enable plugin content:
[image]

Click here:
[image]

And double click the BP_FLyingPawn:
[image]

Notice these rotation components:
[image]

and the SetupPropRotationMovement function that wires up the Rotation components to the actual 3D static meshes for each prop:

[image]

So notice for this to work each prop has to be it's own separate mesh 

[image]

so notice the drone is a compound mesh:

[image]

sytelus, 2017-04-29:
One other change you would need to do is positioning 6 rotors in same place that your frame has (i.e. select coordinate and rotation directions). This is being done in MitiRotorParams::initializeRotorQuadX

liuli2008212, 2017-05-01:
OK,I'm changing as the way of you and lovettchris,thank you for your help.

lovettchris, 2017-05-01:
Good luck, let us know how it goes !

closed 2017-05-01 by lovettchris