microsoft/AirSim #185: DroneShell can't continue the next step

lxgen, 2017-04-27:
Firstly, following the illustration, I start a ue4 project. However when I start the DroneShell,  the Console always shows that " Waiting for drone to report a valid GPS location ...... " and I can't type anything. Does It seem that the DroneShell can't connect to DroneServer? How should I configure the "settings.json" file? Thank you. 
It seems like this:
==||=>
        Welcome to DroneShell 1.0.
        Type ? for help.
        Microsoft Research (c) 2016.

Waiting for drone to report a valid GPS location.....................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................

Thusheep, 2017-04-27:
Had the same problem. Make sure you are connected to your Flight Controller like PX4. If you want to use SITL, follow those instructions:
https://github.com/Microsoft/AirSim/blob/master/docs/sitl.md

The settings.json file should look like this as approved in the demo instructions:
https://github.com/Microsoft/AirSim/blob/master/docs/unreal_demo.md

{
  "Pixhawk": {
    "LocalHostIp": "127.0.0.1",
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14388,
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": true,
    "VehicleCompID": 1,
    "VehicleSysID": 135
  }
}

Then it should work. Hope this helped.

lovettchris, 2017-04-27:
Also check this link for the latest on the settings file format:
[https://github.com/Microsoft/AirSim/blob/master/docs/settings.md](https://github.com/Microsoft/AirSim/blob/master/docs/settings.md)

sytelus, 2017-04-29:
Please re-open if problem still continues.

closed 2017-04-29 by sytelus