microsoft/AirSim #186: Compilation errors: C2100 andC2679 illegal indirection and  '+=': no operator found

FlyingRobotITU, 2017-04-27:
I am using Visual Studio 2015 update 1 and upon building having following errors.

1. `c:\airsim\airsim\airlib\include\common\common_utils\json.hpp(886): error C2100: illegal indirection[C:\AirSim\AirSim\AirLib\AirLib.vcxproj]`
2. c:\program files (x86)\microsoft visual studio 14.0\vc\include\chrono(769): error C2679: binary '+=': no operator found
 which takes a right-hand operand of type 'const std::chrono::duration<double,std::ratio<1,1>>' (or there is no accepta
[Build error.txt](https://github.com/Microsoft/AirSim/files/961938/Build.error.txt)
[image]

ble conversion) [C:\AirSim\AirSim\AirLib\AirLib.vcxproj]

lovettchris, 2017-04-27:
I'm using Update 3, I wonder if they fixed some C++ compiler bugs in Update 3 ?

sytelus, 2017-04-29:
Update 1 is not supported. I also doubt if it can compile rpclib even if you go past these error.s

closed 2017-04-29 by sytelus