microsoft/AirSim #187: DroneShell commands not working properly

Thusheep, 2017-04-27:
I encountered several problems using DroneShell. Fistly, when using the "takeoff" command the drone doesn't take off properly as shown in the AirSimDemo Video (https://www.youtube.com/watch?v=HNWdYrtw3f0), but it takes off just a little bit above the ground, rotates around the z-axis and finally stays at that height giving out "using minimum takeoff altitude: 2.5m". In fact the altitude changes from alt=121.59 at the bottom to alt=123.6. When I try "takeoff" again, it tries to correct itself while staying at the same point or it fails and either goes crazy or flies away. 

Another problem is when I try to use for example the "movetoposition" command (e.g to move along the x-axis or up the z-axis), the drone just goes crazy by tilting itself abruptly or unrealistically turning around pitch/roll-axis. Then it crashes or flies away giving me the message "RPC Error: rpclib: function 'moveToPosition' threw an exception. (...) lookahead error is continually increasing so we do not have safe control, aborting moveOnPath operation". Does someone has a solution or had the same problem? Is it possible to remove lookahead in the codes?

Also what seemed strange was whenever I hit the play button in UE4, the rotors of the drone are already rotating without having to "arm" it. But when I want to take off or at least try to move it, I have to "arm" it first.

Furthermore, when trying to land with the "land" command, an error emerges, saying "Drone hasn't reported a landing state." Then it flies away or crashes.

An example picture when failing:
[image]

Would be very happy if someone can get me some answers.

lovettchris, 2017-04-27:
First try removing the camera views, those can slow down the FPS so much that flight goes crazy.  Press "0".  Second, takeoff will not work well if the altitude is still drifting, which can happen very early on as shown below:

Global position: lat=47.6415, lon=-122.14, alt=122.849
==||=> arm
==||=> takeoff
==||=> pos
Loc==||=> al position: x=-0.025743, y=0.00627603, z=0.224974
Global position: lat=47.6415, lon=-122.14, alt=125.345

In the above case takeoff didn't work, it jumped a bit then landed again.  Notice that while it was on the ground the altitude has changed from 122 to 125.  Well the default takeoff altitude is only 3 meters, so that's why it is still on the ground.  You could change the default takeoff altitude to 10 meters and avoid this problem.  That default lives in a method called getTakeoffZ.

But notice, that if I disarm and rearm and try again the following happens:

disarm
==||=> pos
Lo==||=> cal position: x=-0.0502019, y=0.0255267, z=-1.55482
Global position: lat=47.6415, lon=-122.14, alt=127.125
pos
Lo==||=> cal position: x=-0.0532834, y=0.022402, z=-1.82069
Global position: lat=47.6415, lon=-122.14, alt=127.391
pos
Lo==||=> cal position: x=-0.0536345, y=0.0223744, z=-1.88103
Global position: lat=47.6415, lon=-122.14, alt=127.451
pos
L==||=> ocal position: x=-0.0541674, y=0.0207134, z=-1.90378
Global position: lat=47.6415, lon=-122.14, alt=127.474

--> Ok, the altitude looks stable at 127, so let's try it:

arm
==||=> takeoff
==||=> pos
Lo==||=> cal position: x=-0.037124, y=0.0238937, z=-3.90122
Global position: lat=47.6415, lon=-122.14, alt=129.471
==||=> pos
Lo==||=> cal position: x=-0.0379395, y=0.00927103, z=-4.09609
Global position: lat=47.6415, lon=-122.14, alt=129.666
==||=> pos
Lo==||=> cal position: x=-0.0489082, y=0.00391759, z=-4.12483
Global position: lat=47.6415, lon=-122.14, alt=129.695
==||=> pos
Lo==||=> cal position: x=-0.0465968, y=0.00254618, z=-4.10829
Global position: lat=47.6415, lon=-122.14, alt=129.678

This time it is really flying and hovering at the 3 meter mark.

I do also notice a bit of yaw in the takeoff process, and this needs to be debugged.  I suspect it is because the PX4 takeoff command does allow us to pass a target Yaw and we always pass zero, so if the drone is not already pointing north it will try to point north.  Perhaps what we should do instead of pass in the "current yaw" so it just stays pointing in the same direction as it is on the ground...  That should be an easy fix.  In fact I might try that right now :-)

JohnDing1995, 2017-04-28:
About the "movetoposition" command and look ahead error, I have same problem with you. I just removed `lookahead_error_increasing++;` in DronecontrollerBase.cpp.
But I must noticeyou that ,through the airsim will not throw any exception after alter` DronecontrollerBase.cpp` ,but the drone still gone crazy.

Thusheep, 2017-04-28:
@lovettchris. Thank you for the quick response. I changed the default takeoff altitude to 10 meters but unfortunately the problem still exists. When I try what you did this happens:

Global position: lat=47.6415, lon=-122.14, alt=121.439
==||=> arm
==||=> takeoff
==||=> pos
==||=> Local position: x=-0.0466042, y=0.00448658, z=-1.04317
Global position: lat=47.6415, lon=-122.14, alt=122.535
pos
==||=> Local position: x=0.0183662, y=0.00356221, z=-1.029
Global position: lat=47.6415, lon=-122.14, alt=122.521

It jumps a bit. Then lands and afterwards takes off again and hovers a little bit above ground. (Sometimes it even goes crazy just after the "takeoff" command.) Now when disarming, the position gets confused and won't stay at a specific altitude:

pos
==||=> Local position: x=-0.0947815, y=3.51845, z=4.70571
Global position: lat=47.6415, lon=-122.14, alt=116.801
pos
==||=> Local position: x=-6.81824, y=7.18336, z=14.2944
Global position: lat=47.6414, lon=-122.14, alt=107.212
pos
==||=> Local position: x=-20.381, y=14.0502, z=33.1286
Global position: lat=47.6413, lon=-122.14, alt=88.378
pos
Lo==||=> cal position: x=-40.6146, y=23.317, z=69.9199
Global position: lat=47.6411, lon=-122.14, alt=51.705
pos
Local posi==||=> tion: x=-65.3946, y=33.6523, z=116.552
Global position: lat=47.6409, lon=-122.14, alt=4.954
pos
Local position: x===||=> -97.5509, y=45.8445, z=178.032
Global position: lat=47.6406, lon=-122.14, alt=-56.524
pos
Local position: x===||=> -46.3215, y=13.9408, z=70.8558
Global position: lat=47.6411, lon=-122.14, alt=50.65
pos
Lo==||=> cal position: x=-76.1005, y=19.8596, z=114.785
Global position: lat=47.6408, lon=-122.14, alt=6.721
pos
Lo==||=> cal position: x=-119.27, y=26.7006, z=176.355
Global position: lat=47.6404, lon=-122.14, alt=-54.847

Now if I "arm" again, the drone just goes totally crazy, takes off and flies away. Just by typing "arm". 

However once or twice somehow by luck your method worked and the drone properly took off. But when I tried to use "movetoposition" in x or y direction, quickly the lookahead error emerged and drone went crazy (but z-direction worked). Do you have any suggestions?

Now @JohnDing1995. Also thank you for your response. I tried this but it doesn't seem to be the correct solution. The lookahead error came up anyway (after rebuilding) and drone goes crazy.

clovett, 2017-04-28:
Those numbers also show unstable x and y, not just altitude, and this would be a real problem.  We need to debug that first.  It would be interesting to look at the HIL gps mavlink messages being sent to the px4.  If you are on windows you can use our logviewer to do that, it is in the repo, we also have docs on that.

lovettchris, 2017-04-29:
One more tidbit, to tell the drone to hover after a command completes set this PX4 parameter:
````
param set COM_OBL_ACT 1
````
The default is "land".  So if the drone keeps landing all the time, this might be the solution.

Thusheep, 2017-05-02:
Now I can takeoff and the drone hovers at about 1m above ground but when trying move commands like 'movetoposition' it throws following error: 
==||=> Async Error occurred: rpc::timeout: Timeout of 5000ms while calling RPC function 'moveToPosition'

Also the logviewer is not working. This happens when trying to build:
[image]

The alternative "click once installer" isn't working either. It just gives me a Windows error. 

Also how can reenable the 3 views? I somehow removed them. Or where exactly should I press "0"? Thank you.

Timebutt, 2017-06-02:
Hi @Thusheep, I had the same error building the Logviewer project. The problem lies in the difference between using a '.' or ',' as decimal mark. If you are in the US, your compiler defaults to `CultureInfo=en-US` and will have no problem to build. If you are in Europe - like me - and your culture uses a different decimal mark, the build or program execution will fail.

To solve this, I added these two lines of code in the `MeshViewer()` constructor (inserted after line [37](https://github.com/Microsoft/AirSim/blob/01c2f7b047cd02619d55cd6409e7ec577f7535d6/LogViewer/LogViewer/Controls/MeshViewer.xaml.cs#L37)):
```
Thread.CurrentThread.CurrentCulture = new CultureInfo("en-US");
Thread.CurrentThread.CurrentUICulture = new CultureInfo("en-US");
```

You'll need use these packages to get everything up and running:
```
using System.Globalization;
using System.Threading;
```
After that, the solution will build and you'll be able to use the LogViewer.

@clovett, my approach will fix the problems others may have building the LogViewer, but I'm not sure if this is a clean way of fixing the issue at hand? I'm not used to programming in C#, maybe there is a better way to set the `CultureInfo` fixed to `en-US`. If this is ok, I'll post a pull-request.

clovett, 2017-06-05:
thanks for the fix, I think I found a slightly more localized way to do this since it should only be an issue for parsing the loaded mesh data.  See [my fix](https://github.com/Microsoft/AirSim/commit/2caf2713a21aba0ff05679aab610331ade37f3d7).

closed 2017-06-10 by lovettchris

nirmita123, 2017-07-05:
@Thusheep , it looks like I am facing a similar so if you found a solution for this could you please share it here .
Thanks

tklinterg, 2017-07-09:
I've got a similar problem. When trying to use the 'MoveToPosition' in the dronehell or the HelleDrone project, the drone goes crazy. Could it be that the sensors detect an obstacle and try to correct the drone somehow and this sends the drone downwards spiral ? 

I'm using the PX4 in the SITL mode when running the projects.

Any support is much appreciated.

Thanks.