microsoft/AirSim #189: Precompiled binaries storage

jeyvison, 2017-04-28:
Hey guys, i'd like to suggest the binaries to be in github. I know.... ,i know...,  git wasn't made for that. But it works and comparing to other forms of storage we have today(which is **none**) it would be really helpfull for us whom doesn't want to compile the code.

I may suggest amazon S3 too. it's a cheap kind of storage but would involve some costs. maybe some open source mirror  would help too but i dont know exactly how it works.

Well, thanks for the amazing tool :smile:

lovettchris, 2017-04-28:
Yeah, git is not a great place for that, but I did put a copy of LanscapeMountains up on [OneDrive](https://1drv.ms/f/s!AoS-ymEfLuCLo-Vl5wVsj_yv7pcCuA).  But the real trick is getting those bits to be maintained.  What we really need is a build system that spits out new cooked binaries every day...

sytelus, 2017-04-29:
As @lovettchris has suggested, we are exploring various options and planning to release official binaries soon!

closed 2017-04-29 by sytelus

jeyvison, 2017-04-29:
Got it! I'd  suggest to maybe use travis and OneDrive(or google drive) API to host the binaries too.  If you need a hand on it, let me know.

clovett, 2017-04-29:
Yeah, I wonder if Travis supports google api already...  We have Travis running and a checked in .travis.yml file...

jeyvison, 2017-04-30:
[dpl](https://github.com/travis-ci/dpl/) has some providers for deployment. One of them is Github releases. Maybe it worths a shot. if not ,then we could think on write a custom provider.