microsoft/AirSim #192: QGroundControl/DroneShell problem

nibill, 2017-05-01:
As these problems are totally different than the other, I thought it is ok to open a new issue.

I should mention that I have zero drone experience. No RC used.

Setup:
Intel Xeon CPU E5-2620 v4 @2.10GHz
64GB RAM
NVIDIA GeForce 1080
Windows Server 2016 Datacenter (I really hope that's not what causes all the problems).
Pixhawk PX4 -  Firmware 1.5.5

In the QGroundControl-application I have the following error: 
[image]

As for the airframe I have selected the Simulation HIL Quadcopter X as mentioned in the video. My pixhawk is blinking blue and updated to the newest firmware. No logs available in QGroundControl.

When I click play in the engine, it looks like this:
[image]

The drone shell:
[image]

The Pixhawk changes to white/purple-ish blinking, in case that matters. Might be something little, but I really have no clue.

Thanks!

lovettchris, 2017-05-01:
yes, I just found a bug in our gps code, it is not sending enough gps messages to PX4, we are working on a fix, stay tuned.

nibill, 2017-05-01:
alright, thanks. I'll wait for a fix then.

As for the QGroundControl, do I even have to care about having some red? I only use the simulations...

And should I worry that I don't see a green "Loading settings from..." after I click play in the engine?

Thank you

lovettchris, 2017-05-01:
For QGC you need to select this airframe, that will stop it complaining about uncalibrated sensors:
[image]

lovettchris, 2017-05-02:
The [issue with gps messages ](https://github.com/Microsoft/AirSim/commit/7c18f3548e1b025c29f8b174764993c7665c6619) is now fixed.  Please try again and reopen this issue if you still have a problem, thanks.

closed 2017-05-02 by lovettchris

nibill, 2017-05-03:
Thanks for the quick fix. I've created a whole new project, but now I'm stuck here:
[image]

In the engine it's fine now, I'd say
[image]

As for QGroundControl, I have selected the correct airframe, but it still complains about Radio and Flight Mode
[image]

Any ideas`? Thank you

lovettchris, 2017-05-03:
So this time I see DroneShell is not getting past "Contacting DroneServer" this is probably because you need to start DroneShell AFTER you start DroneServer (in this case the server is in the Unreal app).    Unfortunately, it currently does not have any retry logic, so if it doesn't find the server it gets stuck, that is on the list of things to fix, but for now just restart DroneShell.    So this means every time you click Stop/Start in the Unreal Editor you need to restart the DroneShell app.

lovettchris, 2017-05-03:
As for the Radio calibration step, if you plan to use a Remote Control radio as I show in [this video](https://www.youtube.com/watch?v=HNWdYrtw3f0&feature=youtu.be), then you should calibrate the radio with your pixhawk using QGroundControl.  You can also set some handy flight modes using a switch on your radio.  But you can use AirSim without a radio and ignore the above red dots in QGC.

nibill, 2017-05-07:
I fixed it now.
In case anyone having the same/similar problem: I used to have my project split up over two partitions. AirSim.sln was on C:\ while my project was on D:\. I guess they couldn't see each other. I created a new project on C:\, and now I can reach the DroneServer. That's all I've changed.