microsoft/AirSim #194: Accessing capture components from within code when HUD sub views are disabled

saihv, 2017-05-02:
For better performance, I am experimenting how things work if the depth, seg and scene sub views are turned off within the AirSim HUD (the draw thread is much faster with these off). Upon first looks, it seems that the capture components within the code are returning NULL when these sub-views are turned off, which in turn affects camera recording etc. : is it possible to keep these views disabled and yet be able to access the capture components/images from inside the code?

bretStaz, 2017-05-26:
The performance gains you are seeing are most likely not from turning off the rendering of the additional views, but instead the turning off of the generation of the data that makes up those views so even if you generate the data but do not display them you will probably see the same rough performance as you see when displaying them to the screen.   So this direction most likely will not yield performance gains you are looking for.

bretStaz, 2017-05-26:
My current guess is the segmentation camera is the most expensive to calculate, I want to have depth camera views for all six sides ideally and not the segmentation.  I will probably start to work on that and let you know what the performance is like without the segmentation camera. 

The Camera Director in the Unreal plugin has a "bool togglePIPSeg();" function that does not look like it is exposed to the client side but exposing each camera toggle to the client would achieve let someone turn off the segmentation camera and keep the others.  

Depth data is probably expensive to calculate (unless they are using hardware ) which is why the viewport for the depth data is smaller then the visual data.

bretStaz, 2017-05-26:
finally found your other thread about depth and segmentation being done via shaders so that answers a lot of my questions about performance impact on segmentation.  Have your improvement for multi-thread logging been merged into master, those changes are appealing.

Dynode, 2017-05-26:
@bretStaz They're added.

saihv, 2017-05-27:
@bretStaz  Thanks for the insight! Quick question: you mentioned that disabling the views would disable the generation of data entirely, but if I remember correctly, even when I pressed 0 to disable the sub-views, I was able to double click on the FpvRenderTarget (or any other render target) within Unreal and still be able to see the stream of camera video.  Have you ever tried something like this? I am going to check this again though because it was quite some time ago and I may be mistaken.

saihv, 2017-05-30:
@bretStaz Yep, can confirm: the generation is not disabled when the subviews are turned off. I can still double click on the render targets and get the streaming video.

bretStaz, 2017-05-30:
Thank you for validation.   :-)

closed 2017-06-10 by lovettchris