microsoft/AirSim #196: Internal GPS Noise From Simulator When Connected to Mavros

exploke, 2017-05-03:
I am attempting to run the drone simulator through a program on another laptop connected by mavros. There is a stable connection, and I am able to arm, takeoff, and somewhat fly the drone through the program, but mavros and the program are receiving a skewed GPS location that is fluctuating between a range of points from the actual GPS location which messes with the program. Is there anyway I can have the simulator give a more exact and constant GPS location?

lovettchris, 2017-05-03:
Can you provide a [log file ](https://github.com/Microsoft/AirSim/blob/master/docs/logging.md) so we can see how much it is fluctuating?  Some amount of noise is being simulated, but it should be relatively small.

exploke, 2017-05-05:
[logs.zip](https://github.com/Microsoft/AirSim/files/979774/logs.zip)

I inputted the 'pos', 'arm' , 'takeoff 10', 'disarm', 'x' commands in that order for this log file.

lovettchris, 2017-05-05:
I'm definitely seeing something funky with lat/lon, it seems to be oscillating out of control.  I need one more log file though.  Can you grab the output to port 14389 ? This will capture the HIL GPS simulated values as well.  And make sure you are using the latest bits:
`
    MavLinkTest -server:127.0.0.1:14389 -logdir:d:\temp
`
[image]

exploke, 2017-05-05:
Sorry Chris I am getting this error "parameter name 'CBRK_USB_CHK' was not found" when I type arm while trying to log. Would you have any idea as to why I am running into this problem?

lovettchris, 2017-05-05:
That usually means you need to restart mavlinktest.  You have to start it after you hit Play.

exploke, 2017-05-05:
I was unable to get it to run with port 14389. I could obtain the pos of the quad but could not arm it. I am getting the same error as above. I ran it on port 14388 if that helps at all. I am attaching the files here.
[logsPort14388.zip](https://github.com/Microsoft/AirSim/files/980077/logsPort14388.zip)

exploke, 2017-05-05:
Also a little Off Topic, but would you have any idea what the focal length of the front Camera view in air sim is?

lovettchris, 2017-05-05:
> I was unable to get it to run with port 14389. I could obtain the pos of the quad but could not arm it. I am getting the same error as above. I ran it on port 14388 if that helps at all. I am attaching the files here.
logsPort14388.zip

Oh, you can run two separate instances of mavlinktest.  One on port 14550 to control it and the other on 14389 to get the logs.

You can find the camera properties in the Unreal Editor, if you open BP_PIPCamera, and select CameraComponent you will see this:
[image]

exploke, 2017-05-05:
I am attaching the log files from ports 14550 and 14389. They are all in the attached folder.

[LogsPort14389.zip](https://github.com/Microsoft/AirSim/files/980374/LogsPort14389.zip)

lovettchris, 2017-05-05:
Thanks, looking at the hilgps in the LogViewer I can see it is oscillating also by about this much:

[image]

[image]

I'm not sure why it would be doing this, I'll have to debug some more...

vatsagandhi, 2017-06-05:
Chris how is airsim simulating gps noise? Is it possible to turn that off?

clovett, 2017-06-05:
It is done in ~\AirSim\AirLib\include\sensors\gps\GpsSimple.hpp,

vatsagandhi, 2017-06-06:
Which lines do we change to rid of the GPS Noise? Would we be messing with the parameters in GpsSimpleParams.hpp or just modifying the GpsSimple.hpp?

exploke, 2017-06-08:
I attempted to change the values of GpsSimpleParams.hpp and the noise parameters on QGroundControl with no solution to the GPS noise. When I play my unreal environment as the drone is in disarm mode, QGroundControl still shows the drone fluctuating sporadically because of the GPS noise. I was thinking it might be a Pix

lovettchris, 2017-06-10:
I synced latest PX4 master branch and it works really well, best yet.  So I attached that version here.  One of the things I'm noticing in this version of PX4 is that there is no longer any IMU slippage after performing a long banked turn.

[px4fmu-v2_default.zip](https://github.com/Microsoft/AirSim/files/1065485/px4fmu-v2_default.zip)

vatsagandhi, 2017-06-10:
I rebuild the plugins folder with the latest git repo and also am using the firmware you have posted above. It seems that the quad isn't staying stable in attitude mode either and position seems to be fluctuating. I am attaching the log file in this comment. I ran into this issue when I was using the old firmware as well. 
[MavLinkLogs.zip](https://github.com/Microsoft/AirSim/files/1066111/MavLinkLogs.zip)

vatsagandhi, 2017-06-10:
I am also attaching the logs for HilGPS. The zipped folder contains logs from port 14550 and 14389.
[HilGPSLogs.zip](https://github.com/Microsoft/AirSim/files/1066122/HilGPSLogs.zip)

lovettchris, 2017-06-12:
Sensor data looks good, but there is a weird oscillation in the z-acceleration that we should look into.  THe other thing you should check is your computer performance, CPU/GPU load.  See [performance info](https://github.com/Microsoft/AirSim/blob/master/docs/hard_drive.md).

[image]

vatsagandhi, 2017-06-12:
The computer seems to be fine. Attached images show the cpu usage and disc. We are also using a Nvidia GTX 1070
[image]
[image]

lovettchris, 2017-06-12:
Yep, that looks fine.  I haven't tried the 1070, but it should be fine also.  If you type the backquote key (`), in AirSim it brings up the Unreal Console where you can type "StartFpsChart", do some flying then "StopFpsChart", this should generate an excel chart of render times.  This should tell us for sure if the 1070 is keeping up.

exploke, 2017-06-13:
Here is the excel chart of the rendering times.

[UEDPIE_0_Demo_Map-FPS-Windows.xlsx.zip](https://github.com/Microsoft/AirSim/files/1071926/UEDPIE_0_Demo_Map-FPS-Windows.xlsx.zip)

lovettchris, 2017-06-13:
The times look great, around 16ms, so this should not be the issue.  There is one big spike to 70ms but perhaps something unusual happened at that point.  So from CPU to GPU everything should be working great, so I cannot figure out from the data you have provided what the core issue is here.  You should be getting controllable smooth flight.  Your original post did mention ROS, have you eliminated that variable?

[image]

exploke, 2017-06-13:
For the the FPS Chart test, I used the RC controller. I am also using a custom ROS program to test the position controls in Airsim, but I get the same problem with z-acceleration oscillations as I do when using the RC controller. When I go into position control mode using the custom ROS program, the drone does not stay stable while in the air; I used the ROS program in Gazebo with no problems.

vatsagandhi, 2017-06-13:
The link to this video shows the issue we are running into very clearly. For the attitude control, we experience something similar with RC controller as well and for drone shell it takes off properly but when I issue the land command the quad starts flying everywhere and is constantly tumbling. 

https://youtu.be/yP6Lp4TN5Ek

lovettchris, 2017-06-14:
Thanks, that is helpful.  Let's try and cut out the Ubuntu/mavros connection and control everything on the laptop.  Your radio is plugged into Pixhawk, AirSim is running on the laptop, I assume the Pixhawk is plugged into the laptop.  So I'm not sure what mavros was doing?

Also, after that kind of bad behavior the Pixhawk attitude estimator gets very confused and takes a while to settle, so what I'd like to see is a clean reboot of Pixhawk.  Run AirSim.  Use the Radio only to arm the drone, then takeoff/land, no mavros.    I'm wondering if mavros is adding some sort of bad round trip latency that is messing things up.

One more thing that is helpful is to run the [LogViewer ](https://github.com/Microsoft/AirSim/blob/master/docs/log_viewer.md) on the laptop, when it is connected over socket it will show the "real" drone attitude as reported by the Pixhawk AttitudeEstimator, and when this bad behavior is happening you will probably see it tilting a lot when the drone is simply sitting on the ground - this is the clue that estimator is out of whack.  If you want until drone is level in LogViewer, then manual flight will typically work a lot better.  

This is what you should be getting.  See [demo video](https://youtu.be/d_FyjKDWQfc).

vatsagandhi, 2017-06-14:
I did a clean reboot of Pixhawk, and am only using the Radio to arm the drone. The drone is flying much more smoothly  and consistently than it was before. The only problem now is I am getting a REJECT AUTO RTL error. I will arm the drone and takeoff, but the screen will prompt the REJECT AUTO RTL error at random times, and I will lose control of the drone. I have also already set the NAV_RCL_ACT to disabled.

[image]
[image]

lovettchris, 2017-06-14:
According to [PX4 GitHub Commander.cpp, line 3448](https://github.com/PX4/Firmware/blob/a1a0dd37dc11ca51476b5f9b9942d3952081d100/src/modules/commander/commander.cpp#L3448) this message happens when the AUTO_RTL state transition has been triggered, but the transition was denied, and this can only happen when [state_machine_helper.cpp, line 422](https://github.com/PX4/Firmware/blob/a1a0dd37dc11ca51476b5f9b9942d3952081d100/src/modules/commander/state_machine_helper.cpp#L422) decides that the global position is invalid, or there is no home position set.  That can happen if the GPS is not settled (no green flashing light on Pixhawk).

Try this:
1. reboot pixhawk
2. run AirSim
3. wait for pixhawk green led to flash showing GPS lock has occurred
4. now try and fly manually.

But all the GPS problems you initially reported on this thread could cause the GPS lock to be lost, and that would also cause this.  Question is why was RTL triggered, did QGroundControl do it?  Do you have a GeoFence enabled under Safety tab ?

closed 2017-07-12 by exploke