microsoft/AirSim #199: Modified my Hexrcopter,but can't takeoff

liuli2008212, 2017-05-04:
I have modified as https://github.com/Microsoft/AirSim/issues/184 .Show as below:
 1. Function initializeRotorQuadX from MultiRotorParams.hpp as below:
 if (rotor_count == 6) {
            rotor_poses.clear();
            /* Note: rotor_poses are built in this order: rotor 0 is CW
             (2) |  (4)
                  |
         (1)-------(0)
                  |
             (5) |  (3)
            */
            Quaternionr quadx_rot(AngleAxisr(M_PIf / 6, unit_z));
            rotor_poses.emplace_back(VectorMath::rotateVector(Vector3r( 0,-arm_lengths[1], rotor_z), quadx_rot, true),
                unit_z, RotorTurningDirection::RotorTurningDirectionCCW);
            rotor_poses.emplace_back(VectorMath::rotateVector(Vector3r( 0,-arm_lengths[3], rotor_z), quadx_rot, true),
                unit_z, RotorTurningDirection::RotorTurningDirectionCCW);
            rotor_poses.emplace_back(VectorMath::rotateVector(Vector3r( 0, arm_lengths[4], rotor_z), quadx_rot, true),
                unit_z, RotorTurningDirection::RotorTurningDirectionCCW);
            rotor_poses.emplace_back(VectorMath::rotateVector(Vector3r( arm_lengths[0], 0, rotor_z), quadx_rot, true),
                unit_z, RotorTurningDirection::RotorTurningDirectionCW);
            rotor_poses.emplace_back(VectorMath::rotateVector(Vector3r( arm_lengths[2], 0, rotor_z), quadx_rot, true),
                unit_z, RotorTurningDirection::RotorTurningDirectionCW);
            rotor_poses.emplace_back(VectorMath::rotateVector(Vector3r( -arm_lengths[5],0,  rotor_z), quadx_rot, true),
                unit_z, RotorTurningDirection::RotorTurningDirectionCW);
        }
2. Function setupFrameGeneric from Px4QuadX.hpp  as below:
 //set up arm lengths
        params.rotor_count = 6;
        std::vector<real_T> arm_lengths(params.rotor_count, 0.800f); 
        //set up mass
        params.mass = 28.0f; //can be varied from 0.800 to 1.600 
        real_T motor_assembly_weight = 1.06f;  //weight for MT2212 motor for F450 frame 
        real_T box_mass = params.mass - params.rotor_count * motor_assembly_weight;

        // using rotor_param default, but if you want to change any of the rotor_params, call calculateMaxThrust() to recompute the max_thrust
        // given new thrust coefficients, motor max_rpm and propeller diameter.
        params.rotor_params.calculateMaxThrust();

        //set up dimensions of core body box or abdomen (not including arms).
        params.body_box.x = 0.260f; params.body_box.y = 0.260f; params.body_box.z = 0.040f;
        real_T rotor_z = 2.5f / 100;

        //computer rotor poses
        initializeRotorQuadX(params.rotor_poses, params.rotor_count, arm_lengths.data(), rotor_z);

        //compute inertia matrix
        computeInertiaMatrix(params.inertia, params.body_box, params.rotor_poses, box_mass, motor_assembly_weight);
3. Struct RotorParams from RotorParams.hpp as below:
real_T C_T = 0.002500f; // the thrust co-efficient @ 6396.667 RPM, measured by UIUC.
            real_T C_P = 0.079000f; // the torque co-efficient at @ 6396.667 RPM, measured by UIUC.
            real_T air_density = 1.225f; //  kg/m^3
            real_T max_rpm = 4000.0f; // revolutions per minute
            real_T propeller_diameter = 0.7110f;   //diameter in meters, default is for DJI Phantom 2 
            real_T propeller_height = 0.203f;   //height of cylinderical area when propeller rotates, 1 cm 
            real_T control_signal_filter_tc = 0.005f;    //time constant for low pass filter

            real_T revolutions_per_second;
            real_T max_speed; // in radians per second
            real_T max_speed_square;
            real_T max_thrust = 137.2000f; //computed from above formula for the given constants
            real_T max_torque = 491.0306f; //computed from above formula

			// set this to zero for more accurate simulation to match playback from real drone.
			real_T throttle_boost = 0.2f; // boost the normal throttle control signal to make it more fun to fly
4. BP_FLyingPawn:
[image]
I modified my hexrcopter as above. But only motor 3 and 2 is spinning,other motors cann't spin.The hexrcopter can't takeoff. Please check my modification,thank you.

lovettchris, 2017-05-04:
Mass of 28 seems high, the units are kilograms.  C_T seems low, our default was 0.109919, ideally C_T was measured at the max_rpm you specified at 4000 (so your C_T comment needs updating).  Prop diameter of 71 centimeters is big.  Prop_height is the pitch or twist you see on the prop blade, 20 cm would make your blades into an industrial fan... Think of it this way, while prop is spinning, how tall is the cylinder created by the prop itself.  Probably more like 1-2 cm.

You also need to change this constant in FlyingPawn.h:
static constexpr size_t rotor_count = 6;

I also realized I missed a step before, you will also need PX4 to be in hexacopter mode so it generates output for all 6 motors.  But we haven't tested running simulator with other airframes, and there seems to be some problems there which is why they have "HIL Quadrocopter X" airframe designed for HIL mode simulation.  But what we need then is a "HIL Hexacopter X" airframe configuration.  So to add this new airframe you need to add it to PX4 source code, and rebuild the firmware, then add it to QGroundControl so you can select it.  If you want to try that I can give you some pointers, let me know.

Ideally we should fix PX4 so that the "HIL" airframes are not required, but that might be more work.

lovettchris, 2017-05-04:
I found one more place that needs fixing.  The MultiRotorConnector was also hard coded to 4 rotors:

    real_T rotor_speeds_[4];
    int rotor_directions_[4];
    real_T rotor_thrusts_[4];
    real_T rotor_controls_filtered_[4];

I just checked in a fix so that MultiRotorConnector uses whatever rotor_count is passed in via MultiRotorParams.  I also added the generic Hexacopter config as per your code above, the "Model" is called "Hexacopter".  I have not added any mesh though, so you will still need to do that.

But I slapped together a funky mesh with a couple more props to prove it can work, see attached:

[Hex.zip](https://github.com/Microsoft/AirSim/files/977517/Hex.zip)

I did have to add the HIL hex airframe also.  I will write up how I did it and post it to our docs folder.

liuli2008212, 2017-05-05:
lovettchris,
  Thank you for your step-by-step modification.I will modify with your way,and check the HIL mode simulation about PX4. I found that I have more unknown questions,we will discuss in-depth.Thank you.

lovettchris, 2017-05-05:
I got something better working, [more details here](https://github.com/Microsoft/AirSim/blob/master/docs/hexacopter.md).

closed 2017-05-05 by lovettchris

mchowdhury-tft, 2019-05-20:
> I got something better working, [more details here](https://github.com/Microsoft/AirSim/blob/master/docs/hexacopter.md).

The link is not working. Could you kindly post it here?

lovettchris, 2019-05-20:
Right sorry, correct link is https://github.com/microsoft/AirSim/pull/1890

mchowdhury-tft, 2019-05-20:
Thanks a lot. 
I having hard time on attaching the propeller with the body. I have two .ob file I imported on unreal engine editor such as DJI_Motor_Prop, DJI_S900. I created a Prop0...Prop5 in the blueprint, I dragging the prop to the body and trying to attach it with the body, which does not work once i Play. 
Please let me know about this....