microsoft/AirSim #204: code structure

fmzhu, 2017-05-08:
I'm trying to add the new function " Error Injection" for sensors using the Socket. But the sensors are header-only models for Barometer, IMU, GPS and Magnetometer, so how the drone in the simulator read and get the sensor data from the four sensors. Are the sensors in the SITL-Mode using the sensors from the PX4 instead of the sensors from the AirSim itself?

lovettchris, 2017-05-12:
When using a PX4 based drone (either HIL or SITL) the sensor data is gathered up from the header only models and sent via MAVLINK message to PX4, you will see this code in MavLinkDroneController::sendHILSensor and MavLinkDroneController::sendHILGps.  Similarly, the data coming from PX4 is gathered up by MavLinkVehicle into VehicleState structure.  MavLinkDroneController gets that state in updateState(), which is used to implement getPosition, getVelocity, getHomePoint, getOrientation and so on and those methods are used by the simiulator to show where the drone is and how it is oriented.  

So you could inject errors at the MAVLINK stream socket level by inserting yourself inbetween PX4 and AirSim.  You can do this today using MavLinkTest as a mavlink proxy service like this:

`MavLinkTest -serial:*,115200 -proxy:127.0.0.1:14560
`
AirSim will connect to port 14560 if you set the ~/Documents/AirSim/settings.json "UseSerial" setting to false.  Now every mavlink message is being sent through MavLinkTest, so you could mess with those mavlink messages in either direction to inject errors.

fmzhu, 2017-05-17:
@lovettchris thank you for your kindly help! Now I'm doing my master thesis about error injection. and I'm very interested in the AirSim, because regarding of the AI  AirSim is much better than the Gazebo+ROS. I'm not familiar with c++ and the AirSim is very new and before I concentrate on the development of hardware.  The time is very limited. So would you please give me an easy example(the suggestion above you mentioned) for the "Error Injection", so that I can have a better understanding. I'm going to plan to develop a QT_Socket_GUI, which is used for the end users to inject errors.  I really appreciate your kindly help! 
My E-mail: fmzhu.de@gmail.com

lovettchris, 2017-05-17:
You're welcome.  If you want write a QT based GUI to control inject error injection over a socket then I would still recommend the approach I outlined earlier in this thread.  Look at the MavLinkTest app includes in the AirSim source code.  You can copy most of that code to your QT based GUI so that it can talk to your Pixhawk hardware over serial port, and funnel all mavlink messages to/from AirSim over socket 127.0.0.1:14560.  Then your error injection can simply mess with various mavlink messages.

fmzhu, 2017-05-17:
@lovettchris thank you very much for your reply. So it works also for the SITL mode?

lovettchris, 2017-05-17:
yep

fmzhu, 2017-05-19:
@lovettchris hello, I used MavLinkTest as a mavlink proxy service like this as your suggestion:

MavLinkTest -serial:*,115200 -proxy:127.0.0.1:14560

But the default proxy_port is like this:  #define DEFAULT_PROXY_PORT 14580, should I change the Port_Nummer to 14580? Thank you very much!

closed 2017-05-25 by lovettchris

lovettchris, 2017-06-12:
Ah, yes the SITL ports are different, and a bit more complicated because SITL requires "two separate UDP channels".  One channel sends AirSim hilsensor messages and gets the PX4 output which is "UdpPort": 14560, and the other channel is how AirSim sends flight commands, which is "SitlPort": 14556.  You see this in your ~/Documents/AirSim/settings.json file:

    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,

So to intercept the PX4 stream from SITL you would need MavLinkTest to grab the SitlPort 14560 published by SITL and redirect it to another port, then change your settings.json "SitlPort" setting so that AirSim picks up the new port from MavLinkTest.    So you would do this:

MavLinkTest -udp:127.0.0.1:14560 -proxy:127.0.0.1:14561

Then change settings to this: "UdpPort": 14561.

Similarly, to intercept the "control" stream from AirSim back to PX4, you would need to intercept the output from AirSim and send it to the PX4 SITL port 14556, another instance of MavLinkTest could do that as follows:

MavLinkTest -server:127.0.0.1:14777 -proxy:127.0.0.1:14556

changing settings to:  "SitlPort": 14777.

Something like this should work, but you may have to fix some issues.  For example, we don't want MavLinkTest generating any heartbeat messages in this scenario, so you may need to make that optional.