microsoft/AirSim #205: Low FPS when recording scene

cca2016, 2017-05-09:
When I record the segment or depth image, the speed is very fast and has a high FPS 
however when it comes to the scene, FPS is too low to record and the game cannot play. Why? How to improve?
(PS: I use debuggame editor)

Timebutt, 2017-05-22:
HI @cca2016, this is a known issue and is reported as a bug in #89.
Seems the AirSim rendering thread is interrupted too much by the recording to maintain a stable high frame rate.

The way I currently capture video from AirSim simulations, is through the wonderful OBS Studio software. Mainly targeted at generating live stream video, it also serves beautifully as screen capture software. You can define the screen area you want to capture, and use keyboard shortcuts to start/stop the recording (very nifty, you won't have to click around all the time).

@lovettchris, the AirSim documentation doesn't clearly mention yet how we can efficiently gather training data. I was thinking about detailing the procedure I currently use successfully in either a blog post or GitHub documentation, do you guys have anything planned regarding this matter?

saihv, 2017-05-25:
There was a recent addition to the AirSim code that moves the camera recording to a separate thread and thus does not affect the game thread. (https://github.com/Microsoft/AirSim/pull/162)  

Do you still see FPS drops with this feature?

cca2016, 2017-06-06:
@saihv I have tried the latest version of AirSim. However, I cannot find the three cameras any more. How to fix it?

cca2016, 2017-06-06:
@saihv I have found cameras and the condition is improved and FPS is higher. However, when I am recording the scene image, the drone shaked greatly and is not smooth. The image I get isn't very ideal.
How do that happen?

saihv, 2017-06-06:
I'm sorry, I don't understand the shaking issue. Were you piloting the drone as you were recording? Do you have a PIXHAWK connected? If the drone is being unstable while using PIXHAWK HIL, may be you need some PID tuning on the PIXHAWK.

lovettchris, 2017-06-10:
We've done some work on the camera api, see [camera_views.md](https://github.com/Microsoft/AirSim/blob/master/docs/camera_views.md).  You should now be able call getImageTypeForCamera without having any stuttering effect on main view.

closed 2017-06-10 by lovettchris