microsoft/AirSim #206: Views missing

Thusheep, 2017-05-09:
After pulling the recent Airsim git, the three views (scene, segmentation, depth) disappeared. When pressing the Play button in the unreal editor, only the drone appears. How and where exactly can I enable the views again?

lovettchris, 2017-05-10:
Press "F1" for help, the help will say press "0" to get all 3 views back, or "1", "2" and "3" to select the specific views you want.  We removed them by default because some folks were having performance problems with all 3 views active by default.

Thusheep, 2017-05-10:
Pressing "F1" opens the help from Unreal Engine 4 in the internet explorer. So I cannot access to the help you mean. 
The thing is that I managed to extract the views before as a .png file and now I get an error when using "getimage" which says:
==||=> getimage
==||=> error getting image, check sim for error messages

lovettchris, 2017-05-10:
You are probably getting the Unreal Editor help.  Click on the AirSim view first so it receives keystrokes.  Same is true for the "0", "1", "2" and "3" keystrokes.  You should see this:  

[image]

Thusheep, 2017-05-11:
Unfortunately it always opens the Unreal Editor help, even if I click inside the Airsim view.

Thusheep, 2017-05-11:
Can I set them as default in the codes and when yes where can I do that?

lovettchris, 2017-05-11:
Yes, see  Unreal/Plugins/AirSim/Source/PIPCamera.h, line 33, change:

`static constexpr EPIPCameraType DefaultEnabledCameras = EPIPCameraType::PIP_CAMERA_TYPE_NONE;`

to

`static constexpr EPIPCameraType DefaultEnabledCameras = EPIPCameraType::PIP_CAMERA_TYPE_ALL;`

Thusheep, 2017-05-12:
OK. Thank you.

closed 2017-05-12 by Thusheep

Mayankm96, 2017-05-12:
Well I was facing the same issue. You may also use `Shift+ 1 or 2 or 3` to get different views provided by the simulator. :)

Mayankm96, 2017-05-16:
Well when I press `F1` the menu comes and I am able to choose the views. However, my simulator goes to WireFrame view and I can't escape it until I stop the simulation.

lovettchris, 2017-05-16:
Yes, that is an annoying "feature" of Unreal Editor, it won't do this in the "shipping" game.  But you can get out of wireframe mode by pressing "F3".  Although pressing "F1" to hide the help will also do it.