microsoft/AirSim #208: Does Airsim support mission APIs ?

shakthi-prashanth-m, 2017-05-10:
Hi, Warm Greetings!
Does AirSim has options for 
1. Downloading missions from Aircraft
2. Creating local missions
3. Uploading local missions to the Aircraft
?

Thanks,
Shakthi

lovettchris, 2017-05-11:
There are multiple options for flying missions:  

1. If you have a PX4 based flight controller then you can [Plan Missions using QGroundControl](https://dev.px4.io/en/qgc/), and AirSim will fly those missions by simply putting your drone enters auto flight mode, just as it works with a real PX4 based drone.

2. You can write code that calls the DroneControlBase API to fly missions autonomously, for example "moveOnPath" can take a vector<Points> that can be essentially infinite in length if you want to fly really long missions.  Only limited by available memory on your PC.  Your you can break the flight up into multiple calls to moveOnPath in case you want to compute different paths with high level path planning algorithms.

3. Option 1 is "uploading missions to aircraft", option (2) is on separate PC, but if you build a [custom drone like this](https://github.com/Microsoft/AirSim/blob/master/docs/custom_drone.md) then your "PC" can be on the drone, and therefore you can (2) on the drone while it is flying.

shakthi-prashanth-m, 2017-05-11:
Thank you Chris!
Yes, we can do mission planning using QGC..
So, I meant that AirSim::AirLib does not support option 1 yet.

lovettchris, 2017-05-11:
Right, AirLib does not have an abstraction for PX4 missions because AirLib wants to be cross-drone friendly and not all drones support mavlink missions.  Besides, the PX4 implementation is very memory limited in what it can do.  So this is why we recommend the "big brain"/"little brain" architecture where your "mission" can use the richer DroneControlBase API.  

But if you want to expose the mavlink mission API up to AirSim you could change MavLinkVehicle adding mission API, and you could call that from MavLinkDroneController in AirSim, thereby bypassing the DroneControlBase API.  It would take a couple days to grock the MavLink mission messages and wrap those in MavLinkVehicle.  The new API might be 
```
AsyncResult<std::vector<MavLinkMissionItem>> getMissionItems();
AsyncResult<bool> setMissionItems(std::vector<MavLinkMissionItem> mission);
```
The implementation of these would be using the following messages from MavLinkMessages.hpp:
```
MavLinkMissionRequestPartialList
MavLinkMissionWritePartialList
```

This would make a great pull request :-)

shakthi-prashanth-m, 2017-05-13:
Ok.. thanks a lot for Api doc. Sure 😊

closed 2017-05-13 by shakthi-prashanth-m

vikasWTT, 2025-04-08:
@lovettchris @shakthi-prashanth-m  Hey Guys, i am somewhat doing same, but i finding it bit difficult, in MavlinkMultirotorApi class i tried requesting mission request list sending as message on px4 node but its not replying back , can you give me more idea what i can implement in MavlinkVehicle class to get it working or can i create another mavlink node to get the mission data from px4?. thanks :)