microsoft/AirSim #210: Attitude - GPS mode when using SITL simulation

Timebutt, 2017-05-11:
I went looking through the documentation, but can't find how the AirSim drone flight mode can be set to either Attitude or GPS. When I start the _AirSimGameMode_ in Unreal, it defaults to an Attitude mode that does not maintain height (so no altitude mode). Guess this is the default PX4/posix configuration.

So basically two issues:

- How can I switch the default Attitude mode to a GPS mode in a SITL situation? I'm currently using a Phantom 2 controller connected to the posix simulation using QGroundControl.

- How can I have the drone hover in place when I let go of the throttle stick? I have tried adjusting the tuning in QGroundControl (as suggested [here](https://github.com/Microsoft/AirSim/blob/a935a1967be43ede3814cd292cdf890a2b5f381d/docs/prereq.md#using-the-remote-control)) with no luck. The moment I let the stick go back to its center position, the AirSim drone loses height immediately.

Thanks!

lovettchris, 2017-05-11:
I normally wire that to a switch on my RC remote, but you can do it programmatically if you look inside the MavLinkDroneController.cpp, you will see a mav_vehicle_ object of type MavLinkVehicle.  MavLinkVehicle already has setPositionHoldMode(), but you want setAltitudeHold, which would look like this:

```
AsyncResult<bool>  MavLinkVehicleImpl::setAltitudeHoldMode()
{
    return setMode(static_cast<int>(MAV_MODE_FLAG::MAV_MODE_FLAG_CUSTOM_MODE_ENABLED),
        static_cast<int>(PX4_CUSTOM_MAIN_MODE_ALTCTL));
}

If you add this method then you could call it from MavLinkDroneController when you want to enable that.

Timebutt, 2017-05-12:
Perfect! Thank you for the suggestion, will implement this method and programmatically set it. No other way, as my controller limits the options.

closed 2017-05-12 by Timebutt

lovettchris, 2017-05-12:
Great, please include it in a pull request too :-)

andre-nguyen, 2017-05-12:
[image]

@Timebutt See where it says "Manual" in the top middle of the image? If you click that it becomes a dropdown menu where you can select the modes your autopilot supports.