microsoft/AirSim #212: Unable to calibrate the Joystick Controller

Mayankm96, 2017-05-12:
I am not sure whether I am missing something but the documentation seems a bit unclear for me. I am trying to use an emulator for Logitech Rumblepad 2. The controller works fine since I used it to try out a game on the PC. However, when I am trying to calibrate the controller on QGroundControl, I have been unsuccessful with it so far. 

```
Launch QGroundControl and you should see a new Joystick tab under stettings:

<image> 

Now calibrate the radio, and setup some handy button actions. For example, I set mine so that the 'A' button arms the drone, 'B' put it in manual flight mode, 'X' puts it in altitude hold mode and 'Y' puts it in position hold mode. I also prefer the feel of the controller when I check the box labelled "Use exponential curve on roll,pitch, yaw" because this gives me more sensitivity for small movements.]
```

Why do I need to calibrate the radio if I am using a joystick? And more importantly how should I go about it?

andre-nguyen, 2017-05-13:
>Why do I need to calibrate the radio if I am using a joystick? 

Because just like a regular RC, QGC needs to know the min/max of all channels as it will differ from one gamepad to another.

>And more importantly how should I go about it?

Starting from scratch:
1. Launch px4 SITL in a terminal, it will wait for a UDP connection on a certain port.
1. Launch Unreal Editor and press play. Once the quad spawns, a connection should be made to px4 SITL which will unblock the autopilot and start executing.
1. Start QGC (caution tricky)
    1. Click the white/purple Q in the top right to open the settings.
    1. Add a UDP comm link on port 14540 with the localhost ip (127.0.0.1)
    1. Connect using that link. You should see at the top the sattelite count become valid and the battery voltage too. If you don't see this, then stop now because the Mavlink connection is invalid.
    1. Click the gears icon next to the Q icon and you should get a bunch of menus including the joystick configuration menu.
    1. The rest should be pretty straight forward, click calibrate and follow the instructions on the screen.

Flying:
1. Assign a button to arming and disarming.
2. Click on "Manual" at the top right and change the mode to "Position".
3. Arm by pressing the button from step 1.
4. Throttle up to take off.

Mayankm96, 2017-05-15:
Is there a way to build PX4 SITL in Windows 8.1? BashOnWindows seem to require Windows 10 and I have seen the documentation on the PX4 website but it seems to not work. 

I am really sorry for bothering so much with this. Thank you for your help!

lovettchris, 2017-05-15:
On windows 8.1 you would have to install something like Oracle VirtualBox and run Ubuntu in a virtual machine.  VirtualBox is free and works very well.  You can get the Ubuntu 16.04 iso image that VirtualBox can install from Ubuntu.com.

Mayankm96, 2017-05-15:
Hm I see... Well I upgraded the Windows to 10 and got everything setup (PX4 in SITL). I have calibrated my RC Controller (I did away with the joystick, for now, sorry about that). However, when I am running the UE4 simulator and press the play button I get the error messages: 

```
not receiving any messages from HIL, please restart the HIL node and try again
not receiving any messages from SITL, please restart the SITL app and try again
```

Is there anywhere I might be going wrong with the procedure? I have been trying to figure this out since morning and reread the docs but I am unable to figure it out.

lovettchris, 2017-05-15:
Is the px4 sitl app running before you click "Play" in the Unreal Editor ?

`./build_posix_sitl_default/src/firmware/posix/px4 ./posix-configs/SITL/init/ekf2/iris`

Mayankm96, 2017-05-15:
Yeah and I have the following output on the terminal console:
```
px4 starting.

INFO  [dataman] Unkown restart, data manager file 'rootfs/fs/microsd/dataman' size is 11797680 bytes
INFO  [platforms__posix__drivers__ledsim] LED::init
INFO  [platforms__posix__drivers__ledsim] LED::init
INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..
INFO  [simulator] Got initial simulation data, running sim..
INFO  [pwm_out_sim] MODE_8PWM
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14556 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14557 remote port 14540
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [logger] logger started (mode=all)
pxh> INFO  [logger] Start file log
INFO  [logger] Opened log file: rootfs/fs/microsd/log/2017-05-16/02_57_46.ulg
INFO  [tone_alarm] startup
INFO  [mavlink] partner IP: 127.0.0.1
INFO  [lib__ecl] EKF aligned, (pressure height, IMU buf: 17, OBS buf: 16)
```

lovettchris, 2017-05-15:
Ok, let's double check that your ~/Documents/AirSim/settings.json contains this:
```
    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": false,
```
You should get 4 more lines of output from px4 app:

```
INFO  [lib__ecl] EKF GPS checks passed (WGS-84 origin set)
INFO  [lib__ecl] EKF commencing GPS fusion
INFO  [commander] home: 47.6414680, -122.1401672, 116.53
INFO  [tone_alarm] home_set
```

Mayankm96, 2017-05-15:
Oh I am, somehow it didn't get selected earlier. 
```
INFO  [lib__ecl] EKF GPS checks passed (WGS-84 origin set)
INFO  [lib__ecl] EKF commencing GPS fusion
INFO  [commander] home: 47.6414680, -122.1401669, 102.07
INFO  [tone_alarm] home_set
```

And yes my ` settings.json` file has those settings.

Mayankm96, 2017-05-15:
It might be possible that my airframe isn't getting set to HIL Quadcopter X. Every time I do so, after restarting it goes back to DJI Flame Wheel F330. 

Also I was trying to calibrate my joystick but the first step where it asks me to set the throttle to max, I do so with the Logitech Controller but it doesn't progresses. I am using the Xbox360ce emulator.

lovettchris, 2017-05-15:
I think I see the same error, and it is bogus, just ignore it.  There is a "timeout" that was triggered because px4 app was slow to respond.  You should be able to arm the drone  using DroneShell.exe and takeoff.  Can you try that?

Mayankm96, 2017-05-15:
Yes I am able to do the takeoff from the DroneShell.

lovettchris, 2017-05-15:
Great, so ignore the bogus message for now, I'll try and fix the timeouts on that so it doesn't mislead. Thanks for reporting the issue.

Mayankm96, 2017-05-15:
I am still not able to get the HIL working. Any idea on how to fix that?

lovettchris, 2017-05-15:
Ok, back to the original issue...  I take it QGroundControl is not showing this Joystick tab?
[image]

See if any of this [info for xbox controller ](https://github.com/Microsoft/AirSim/blob/master/docs/xbox_controller.md) is useful.

Mayankm96, 2017-05-16:
Well it is showing the joystick under vehicle setup. However, that is not the issue I am facing. When I start calibrating the remote controller ([Logitech Cordless Rumblepad 2](http://support.logitech.com/en_us/product/cordless-rumblepad-2)), I am unable to do it. My controller is working fine because the XBox emulator does work with it.

lovettchris, 2017-05-16:
I wonder if you need x360ce.com app to help emulate xbox controller?

Mayankm96, 2017-05-17:
Well I have tried using that.  Although I am able to see the gamepad working on it, on the QGC nothing happens.

Mayankm96, 2017-05-18:
Well I used an [XBox One controller](https://www.amazon.com/Xbox-One-Wireless-Controller-Without-millimeter/dp/B00CMQTUSS) and it is working fine for now. Still trying to figure out what is going wrong with the Logitech one and the emulator.

Mayankm96, 2017-06-10:
I guess I don't need an answer to this issue anymore since the simulation now works fine for me. 

Thank you for your help!

closed 2017-06-10 by Mayankm96