microsoft/AirSim #214: error during setup "console" for toolchain of px4. How fix it?

AzhariAzali, 2017-05-12:
How to overcome this problem?

**** Build of configuration Default for project Firmware ****
make all
/bin/sh: cmake: command not found
rm: cannot remove directory `/c/px4/Firmware/build_posix_sitl_default': Permission denied
make: *** [posix_sitl_default] Error 1

I have already install cmake but I don't know how to interrupt it.
Hopefully if there is any suggestion for the solution...
tq

Aperture-Laboratories, 2017-05-15:
What version of cmake do you have installed? 
What are the permissions of '../build_posix_sitl_default'?
Which OS are you running?

lovettchris, 2017-05-15:
note: I was getting similar bogus error about missing cmake just now that was fixed by doing "git clean -dfx", the problem was I moved cmake, so the old make files references the wrong location.  The clean allowed the makefiles to be rebuilt and everything is now good.

closed 2017-06-10 by lovettchris