microsoft/AirSim #215: Switching between camera views quickly

hngenc, 2017-05-14:
My code seems to take a significant performance hit when switching between camera views. For example, the following code:

	// 46 FPS
	client.setImageTypeForCamera(0, DroneControllerBase::ImageType::Scene);
	while (1) {
		std::vector<uint8_t> img;
		
		do {
			img = client.getImageForCamera(0, DroneControllerBase::ImageType::Scene);
		} while (img.size() <= 1);
	}

allows me to pull about 46 images per second. However, the code below only pulls about 6 images per second:

	// 6 FPS
	while (1) {
		std::vector<uint8_t> scene;
		std::vector<uint8_t> depth;

		client.setImageTypeForCamera(0, DroneControllerBase::ImageType::Scene);
		do {
			scene = client.getImageForCamera(0, DroneControllerBase::ImageType::Scene);
		} while (scene.size() <= 1);

		client.setImageTypeForCamera(0, DroneControllerBase::ImageType::Depth);
		do {
			depth = client.getImageForCamera(0, DroneControllerBase::ImageType::Depth);
		} while (depth.size() <= 1);
	}

Is there any way to reduce this performance impact? I tried the following code as well, but for some reason, I am only able to pull images from camera 0:

	// Doesn't work
	client.setImageTypeForCamera(0, DroneControllerBase::ImageType::Scene);
	client.setImageTypeForCamera(1, DroneControllerBase::ImageType::Depth);

	while (1) {
		std::vector<uint8_t> scene;
		std::vector<uint8_t> depth;

		do {
			scene = client.getImageForCamera(0, DroneControllerBase::ImageType::Scene);
		} while (scene.size() <= 1);

		do {
			// This code fails to work with camera 1
			depth = client.getImageForCamera(1, DroneControllerBase::ImageType::Depth);
		} while (depth.size() <= 1);
	}
`

saihv, 2017-05-27:
You're getting 46 images per second with the first method? That's really surprising, because the getImageForCamera() performs a direct pixel read which blocks the game thread, and works at about 1-2 Hz only..

lovettchris, 2017-06-02:
This should be fixed with this [pull request](https://github.com/Microsoft/AirSim/pull/255). The get/setImageTypeForCamera API is going away and the performance now will be as good as the render thread can manage to keep up.

hngenc, 2017-06-05:
@saihv My code would repeatedly poll the camera until it returned an image of size greater than 1 byte, in order to avoid returning empty data.

Sounds great @lovettchris! I'll close this issue then.

closed 2017-06-05 by hngenc