microsoft/AirSim #217: software architecture of Airsim

jonweili, 2017-05-15:
Hello, I'm new about Airsim, I have read the instruction already. But, I am still infused about the software architecture about Airsim, especially, the main function. I have found several main functions in the whole project, but which is the real main after we have built it?

lovettchris, 2017-05-15:
When running the Unreal simulator, the "Unreal engine" is the main entry point, the unreal engine then calls into AirSim library via plugin interfaces.  The other main functions you found are just for standalone tools like DroneShell, DroneServer and MavLinkTest and the HelloDrone sample app.

jonweili, 2017-05-16:
Thanks lovettchris, but I am still confused about the main function. For example, I want to add code to print  sensors data via serial port. But where should I put the code? And if I want to build the sample app, what should I do?

jonweili, 2017-05-16:
@lovettchris ,I think the following codes is the main function inserting to unreal engine, is this right?
`// AirLibUnitTests.cpp : Defines the entry point for the console application.
//

#include <cassert>
#include "TestVehicles.hpp"

int main()
{
	Settings& settings = Settings::loadJSonFile("settings.json");

	TestVehicles test1;
	test1.Run();
	
    return 0;
}`

lovettchris, 2017-05-16:
No that is just a unit test. AirSim runs on the [Unreal Game engine](https://www.unrealengine.com/what-is-unreal-engine-4) which is a big complex application consisting of a game editor and a game runtime.  The runtime provides the environment in which AirSim executes.  When Unreal loads an environment that has the GameMode set to AirSimGameMode the first line of code that executes inside AirSim is this constructor:
```
void AAirSimGameMode::AAirSimGameMode
```
And this is what the call stack looks like on the MainThread :
````
>	UE4Editor-AirSim-Win64-DebugGame.dll!AAirSimGameMode::AAirSimGameMode(const FObjectInitializer & ObjectInitializer) Line 12	C++
 	UE4Editor-AirSim-Win64-DebugGame.dll!InternalConstructor<class AAirSimGameMode>(class FObjectInitializer const &)	C++
 	UE4Editor-CoreUObject.dll!00007ffa61e5248f()	Unknown
 	UE4Editor-CoreUObject.dll!00007ffa61fea310()	Unknown
 	UE4Editor-CoreUObject.dll!00007ffa61fcc5b0()	Unknown
 	UE4Editor-CoreUObject.dll!00007ffa61e55ab7()	Unknown
 	UE4Editor-Core.dll!00007ffa4f667173()	Unknown
 	UE4Editor-Core.dll!00007ffa4f82c70a()	Unknown
 	UE4Editor-Projects.dll!00007ffa76d1b4f2()	Unknown
 	UE4Editor-Projects.dll!00007ffa76d1a875()	Unknown
 	UE4Editor.exe!00007ff7756b2913()	Unknown
 	UE4Editor.exe!00007ff7756b6a02()	Unknown
 	UE4Editor.exe!00007ff7756ae96a()	Unknown
 	UE4Editor.exe!00007ff7756c0a0a()	Unknown
 	UE4Editor.exe!00007ff7756c22b6()	Unknown
 	kernel32.dll!BaseThreadInitThunk()	Unknown
 	ntdll.dll!RtlUserThreadStart()	Unknown
````
The entire AirSim boot and play cycle looks like this:

[image]

sytelus, 2017-05-25:
The constructor `AAirSimGameMode::AAirSimGameMode` is the kind of `main()` function for AirSim. Here we specify the HUD class. Unreal engine that creates the instance of that class. In case of AirSim that class is ASimHUD and its constructor then creates implementation of ASimMode class.

closed 2017-05-25 by sytelus