microsoft/AirSim #218: Possible problems about [install clang-3.9 in Linux document](https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md)

MiaoDX, 2017-05-15:
I am not so sure of it, maybe it's beacuse the mirrors I use. I tried to install in Linux 14.04, as said at [linux_build.md](https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md), but after installed the signature and updated, clang-3.9 still cannot be located.

After looking at [http://apt.llvm.org/](http://apt.llvm.org/), I add these lines (as shown in the page) to `/etc/apt/sources.list`:

``` vi
deb http://apt.llvm.org/trusty/ llvm-toolchain-trusty main
deb-src http://apt.llvm.org/trusty/ llvm-toolchain-trusty main
# 3.9 
deb http://apt.llvm.org/trusty/ llvm-toolchain-trusty-3.9 main
deb-src http://apt.llvm.org/trusty/ llvm-toolchain-trusty-3.9 main
# 4.0 
deb http://apt.llvm.org/trusty/ llvm-toolchain-trusty-4.0 main
deb-src http://apt.llvm.org/trusty/ llvm-toolchain-trusty-4.0 main
```

And then updated and installed successfully.

Could it be a small mistake (or such a tiny issue and no body cares at all), or I am just wrong and miss something?

lovettchris, 2017-05-15:
You did the right thing, the Linux_build.md does prefix the llvm install with this sentence: "If you are using Ubuntu 16.04 (Xenial) then you can simply do this...".

closed 2017-05-15 by lovettchris

MiaoDX, 2017-05-15:
Aha, gotcha, I was thinking that `For other flavors of Linux` just means `Debian` or `Fedora 20` or `Arch Linux` ...