microsoft/AirSim #219: Xbox controller-based steering of the PIPs while in FPV

Dynode, 2017-05-15:
Proposed concept: Have the PIP views follow the drone while in FPV. The PIPs can look around by using a PS3 / Xbox controller (think typical First Person Shooter mouse-look).

Issues: AirSim seems to support just one active camera, it's not possible to control the drone in FPV and still have the PIPs follow the drone (they switch position with the external camera).

sytelus, 2017-05-16:
Currently we have kept things simple. You have FPV camera and external follower camera. You can swap them in PIP or main view. This design decision is not the most flexible one, of course, but it makes few things simple, for example, no complications to select which camera is PIP or main.

Having said that there are no technical limitations on how many camera you may have and which one is PIP or main. However, as you add more cameras, FPS might start to drop dramatically.

In very near future, the plan is this: We allow to place N PIPCamera objects wherever user like. We can then offer following APIs:

getCameraCount()
getCameraImage(index, imageType)
setCameraToPIPView(index)
setCameraToMainView(index)

Dynode, 2017-05-16:
Thanks. So FPV and the PIPs share the same defined camera? The Depth, Seg. and Scene images stem from the same cam? To have the PIPs steerable by a controller basically means adding new dedicated camera and C++ code to take control over orientation of that cam?

Do you have a rough idea of when your plan / API can be implemented?

closed 2017-05-29 by Dynode

sytelus, 2018-02-21:
I just added Gimbal APIs in AirSim. [Here's doc](https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md#camera-orientation-gimble).