microsoft/AirSim #220: Controller Calibration

eanmikale, 2017-05-16:
Currently the quadcopter Arms via Pixhawk, and I receive all the proper reading from the on-screen display at startup. However, while the quad arms, it will not fly off of the ground. 
[image]

Under project setting, for some reason I am unable to select the default pawn as BP_FlyingPawn with the AirSimGameMode. 
[image]

Here are the default engine inputs. 
[image]

And here is a picture of  the JSON Settings. Any assistance with what I might be doing wrong is greatly appreciated. I have been troubleshooting controls for the past 48 hours. Thank you in advance. 
[image]

I also cannot see any of the other three camera modes. 

Regards, 

EM

lovettchris, 2017-05-16:
Looks like you placed the BP_FlyingPawn on the editor canvas.  You can do that, but it is more complicated, you don't need to do that.  If you delete the pawn and click Play then AirSim will setup the pawn for you.

To setup the pawn yourself, you need to follow the instructions in my [demo video](https://www.youtube.com/watch?v=1oY8Qu5maQQ&feature=youtu.be), skip to 8 minutes 45 seconds where I show how to add the drone manually, with camera and camera director components.

eanmikale, 2017-05-17:
Chris, 

I have been successful at getting the quad to manifest upon pressing PLAY on the Unreal Engine Dashboard. However, while the Drone will power-up when I hold down the sticks, it still will not lift-off. I'm sure it's something simple. Any assistance is appreciated. Thanks again. 

Regards, 

Ean Mikale, J.D.

Chief Innovation Officer

Infinite 8 Institute, L3C

ean@infinite8institute.com

www.infinite8institute.com

U.S. Federal Contractor

---- On Tue, 16 May 2017 14:11:45 -0500 lovettchris &lt;notifications@github.com&gt; wrote ----

Looks like you placed the BP_FlyingPawn on the editor canvas. You can do that, but it is more complicated, you don't need to do that. If you delete the pawn and click Play then AirSim will setup the pawn for you.

To setup the pawn yourself, you need to follow the instructions in my demo video, skip to 8 minutes 45 seconds where I show how to add the drone manually, with camera and camera director components.

lovettchris, 2017-05-17:
Ok, great, getting closer.  I assume you have you calibrated your RC radio with your Pixhawk using QGroundControl.  So let's try this test:

1) find DroneShell.exe (or DroneShell executable if you are on Linux), on my machine it is in D:\git\AirSim\DroneShell\build\x64\Debug\DroneShell.exe
2) Click Play in Unreal then run DroneShell.
3) Wait for DroneShell to report a good GPS lock (Pixhawk takes a few seconds thinking about the simulated GPS signal before it decides that it is good).
4) Then try the following commands:

```
D:\git\AirSim>D:\git\AirSim\DroneShell\build\x64\Debug\DroneShell.exe
==||=>
        Welcome to DroneShell 1.0.
        Type ? for help.
        Microsoft Research (c) 2016.

Waiting for drone to report a valid GPS location............
Global position: lat=47.6415, lon=-122.14, alt=123.796
==||=> pos
==||=> Local position: x=-0.00847796, y=0.00497776, z=10.9747
Global position: lat=47.6415, lon=-122.14, alt=119.528
pos
==||=> Local position: x=-0.00332015, y=0.00573516, z=0.47169

// Notice here that z is still shifting.  Takeoff will not work reliably until z-coordinate stabilizes.

Global position: lat=47.6415, lon=-122.14, alt=129.986
==||=> pos
==||=> Local position: x=-0.00496105, y=0.00559207, z=0.631262
Global position: lat=47.6415, lon=-122.14, alt=129.822
==||=> pos
==||=> Local position: x=-0.00602061, y=0.00496465, z=0.665986
Global position: lat=47.6415, lon=-122.14, alt=129.777

arm  // ok, z-coordinate looks good, let's try taking off.
==||=> takeoff
==||=> pos
==||=> Local position: x=0.145732, y=-0.330749, z=-1.56635
Global position: lat=47.6415, lon=-122.14, alt=132.005

movetoposition -z -10    // it took off, so now let's try moving up higher.
==||=> pos
==||=> Local position: x=0.195341, y=-0.146924, z=-6.86758
Global position: lat=47.6415, lon=-122.14, alt=137.3

==||=> posAsync Error occurred: rpc::timeout: Timeout of 5000ms while calling RPC function 'moveToPosition'

// ignore this error, it is a known issue, you can safely ignore it..

Lo==||=> cal position: x=-0.041718, y=-0.00280605, z=-8.59355
Global position: lat=47.6415, lon=-122.14, alt=139.033

hover  // ok, reached -8 meters, let's switch to hover.

==||=> pos
L==||=> ocal position: x=-0.312998, y=-0.0192368, z=-6.25288
Global position: lat=47.6415, lon=-122.14, alt=136.693

land  // it is hovering, great, now we can land.
==||=>

```

eanmikale, 2017-05-18:
After pressing play inside the engine and executing the DroneShell file, the command screen simply reads, waiting for Drone to report a valid GPS location. I'm assuming this is going off of the artificially created GPS coordinates. Nonetheless, the DroneShell command screen remains in queue, and progresses no further than this point. I am also unable to acquire GPS indoors with the IRIS+.  Were your tests conducted indoors or outdoors, as this information would be helpful. Thanks again. 

---- On Wed, 17 May 2017 15:03:07 -0500 lovettchris &lt;notifications@github.com&gt; wrote ----

Ok, great, getting closer. I assume you have you calibrated your RC radio with your Pixhawk using QGroundControl. So let's try this test:

find DroneShell.exe (or DroneShell executable if you are on Linux), on my machine it is in D:\git\AirSim\DroneShell\build\x64\Debug\DroneShell.exe

Click Play in Unreal then run DroneShell.

Wait for DroneShell to report a good GPS lock (Pixhawk takes a few seconds thinking about the simulated GPS signal before it decides that it is good).

Then try the following commands:

D:\git\AirSim&gt;D:\git\AirSim\DroneShell\build\x64\Debug\DroneShell.exe ==||=&gt; Welcome to DroneShell 1.0. Type ? for help. Microsoft Research (c) 2016. Waiting for drone to report a valid GPS location............ Global position: lat=47.6415, lon=-122.14, alt=123.796 ==||=&gt; pos ==||=&gt; Local position: x=-0.00847796, y=0.00497776, z=10.9747 Global position: lat=47.6415, lon=-122.14, alt=119.528 pos ==||=&gt; Local position: x=-0.00332015, y=0.00573516, z=0.47169 // Notice here that z is still shifting. Takeoff will not work reliably until z-coordinate stabilizes. Global position: lat=47.6415, lon=-122.14, alt=129.986 ==||=&gt; pos ==||=&gt; Local position: x=-0.00496105, y=0.00559207, z=0.631262 Global position: lat=47.6415, lon=-122.14, alt=129.822 ==||=&gt; pos ==||=&gt; Local position: x=-0.00602061, y=0.00496465, z=0.665986 Global position: lat=47.6415, lon=-122.14, alt=129.777 arm // ok, z-coordinate looks good, let's try taking off. ==||=&gt; takeoff ==||=&gt; pos ==||=&gt; Local position: x=0.145732, y=-0.330749, z=-1.56635 Global position: lat=47.6415, lon=-122.14, alt=132.005 movetoposition -z -10 // it took off, so now let's try moving up higher. ==||=&gt; pos ==||=&gt; Local position: x=0.195341, y=-0.146924, z=-6.86758 Global position: lat=47.6415, lon=-122.14, alt=137.3 ==||=&gt; posAsync Error occurred: rpc::timeout: Timeout of 5000ms while calling RPC function 'moveToPosition' // ignore this error, it is a known issue, you can safely ignore it.. Lo==||=&gt; cal position: x=-0.041718, y=-0.00280605, z=-8.59355 Global position: lat=47.6415, lon=-122.14, alt=139.033 hover // ok, reached -8 meters, let's switch to hover. ==||=&gt; pos L==||=&gt; ocal position: x=-0.312998, y=-0.0192368, z=-6.25288 Global position: lat=47.6415, lon=-122.14, alt=136.693 land // it is hovering, great, now we can land. ==||=&gt;

lovettchris, 2017-05-18:
Correct, AirSim is generating GPS, but DroneShell needs to wait until Pixhawk is happy with that GPS signal.  If DroneShell is never reporting this, then that is the problem we need to debug:

`Global position: lat=47.6415, lon=-122.14, alt=123.796
`

it can take about 10 seconds, then you should see the pixhawk green light flash.  If you ware not seeing that then I wonder if you have not selected the "HIL Quadrocopter X" AirFrame using QGroundControl?

eanmikale, 2017-05-18:
Yes, I have correctly selected the HIL Quadcopter X Airframe. Furthermore, the GPS location still does not lock. This is likely where our issue resides. Any ideas? 

---- On Thu, 18 May 2017 00:32:52 -0500 lovettchris &lt;notifications@github.com&gt; wrote ----

Correct, AirSim is generating GPS, but DroneShell needs to wait until Pixhawk is happy with that GPS signal. If DroneShell is never reporting this, then that is the problem we need to debug:

Global position: lat=47.6415, lon=-122.14, alt=123.796

it can take about 10 seconds, then you should see the pixhawk green light flash. If you ware not seeing that then I wonder if you have not selected the "HIL Quadrocopter X" AirFrame using QGroundControl?

lovettchris, 2017-05-18:
Try different versions of Pixhawk Firmware.  I'm using the latest bits from px4 master branch.  But you could also try older versions.  Also see if you can get it working using SITL.  If still no luck please attach some [logging info](https://github.com/Microsoft/AirSim/blob/master/docs/logging.md)

lovettchris, 2017-05-18:
One more thing, if your pixhawk got armed somehow then it will not report home position and this can also cause DroneShell to hang.  If you disarm the drone using radio then DroneShell should be happier.

eanmikale, 2017-05-19:
I did everything you asked a something odd happened. I was never able to get the DroneShell to work. When I press play in the Editor, the scrolling commands tell me that I have GPS, although Drone Shell will not work even with the quad disarmed. For some reason I can now fly, but the left controls for the elevator and the rudder do not work consistently, and I have to possess the quad to fly it in first-person mode. 

Second, the quad tends to tumble as I am flying it, although there are no obstacles near me. I have included the log from Unreal below. Thanks again. 

LogPlatformFile: Not using cached read wrapper

LogInit:Display: Loading recent project file: ../../../../../../Users/Ean/Documents/Unreal Projects/MyProject/MyProject.uproject

LogInit:Display: RandInit(-2145861997) SRandInit(-2145861997).

LogTaskGraph: Started task graph with 5 named threads and 6 total threads with 1 sets of task threads.

LogStats: Stats thread started at 0.059292

LogInit: Using libcurl 7.47.1

LogInit:  - built for x86_64-pc-win32

LogInit:  - supports SSL with OpenSSL/1.0.2g

LogInit:  - supports HTTP deflate (compression) using libz 1.2.8

LogInit:  - other features:

LogInit:      CURL_VERSION_SSL

LogInit:      CURL_VERSION_LIBZ

LogInit:      CURL_VERSION_IPV6

LogInit:      CURL_VERSION_ASYNCHDNS

LogInit:      CURL_VERSION_LARGEFILE

LogInit:      CURL_VERSION_IDN

LogInit:  CurlRequestOptions (configurable via config and command line):

LogInit:  - bVerifyPeer = false  - Libcurl will NOT verify peer certificate

LogInit:  - bUseHttpProxy = false  - Libcurl will NOT use HTTP proxy

LogInit:  - bDontReuseConnections = false  - Libcurl will reuse connections

LogInit:  - CertBundlePath = nullptr  - Libcurl will use whatever was configured at build time.

LogInit: Build: ++UE4+Release-4.15-CL-3416026

LogInit: Engine Version: 4.15.2-3416026+++UE4+Release-4.15

LogInit: Compatible Engine Version: 4.15.0-3299760+++UE4+Release-4.15

LogInit: Net CL: 3299760

LogDevObjectVersion: Number of dev versions registered: 14

LogDevObjectVersion:   Dev-Blueprints (B0D832E4-1F89-4F0D-ACCF-7EB736FD4AA2): 8

LogDevObjectVersion:   Dev-Build (E1C64328-A22C-4D53-A36C-8E866417BD8C): 0

LogDevObjectVersion:   Dev-Core (375EC13C-06E4-48FB-B500-84F0262A717E): 2

LogDevObjectVersion:   Dev-Editor (E4B068ED-F494-42E9-A231-DA0B2E46BB41): 14

LogDevObjectVersion:   Dev-Framework (CFFC743F-43B0-4480-9391-14DF171D2073): 22

LogDevObjectVersion:   Dev-Mobile (B02B49B5-BB20-44E9-A304-32B752E40360): 0

LogDevObjectVersion:   Dev-Networking (A4E4105C-59A1-49B5-A7C5-40C4547EDFEE): 0

[... 452 lines cut ...]
LogWorld: Bringing up level for play took: 0.205748

Cmd: r.CustomDepth 3

r.CustomDepth = "3"

PIE: Info Play in editor start time for /Game/ModularNeighborhoodPack/Maps/UEDPIE_0_Demo_Map 1.909

LogBlueprintUserMessages: Late PlayInEditor Detection: Level '/Game/ModularNeighborhoodPack/Maps/Demo_Map.Demo_Map:PersistentLevel' has LevelScriptBlueprint '/Game/ModularNeighborhoodPack/Maps/Demo_Map.Demo_Map:PersistentLevel.Demo_Map' with GeneratedClass '/Game/ModularNeighborhoodPack/Maps/Demo_Map.Demo_Map_C' with ClassGeneratedBy '/Game/ModularNeighborho

odPack/Maps/Demo_Map.Demo_Map:PersistentLevel.Demo_Map'

LogBlueprintUserMessages: Early EndPlayMap Detection: Level '/Game/ModularNeighborhoodPack/Maps/Demo_Map.Demo_Map:PersistentLevel' has LevelScriptBlueprint '/Game/ModularNeighborhoodPack/Maps/Demo_Map.Demo_Map:PersistentLevel.Demo_Map' with GeneratedClass '/Game/ModularNeighborhoodPack/Maps/Demo_Map.Demo_Map_C' with ClassGeneratedBy '/Game/ModularNeighborhoo

dPack/Maps/Demo_Map.Demo_Map:PersistentLevel.Demo_Map'

LogBlueprintUserMessages: Late EndPlayMap Detection: Level '/Game/ModularNeighborhoodPack/Maps/Demo_Map.Demo_Map:PersistentLevel' has LevelScriptBlueprint '/Game/ModularNeighborhoodPack/Maps/Demo_Map.Demo_Map:PersistentLevel.Demo_Map' with GeneratedClass '/Game/ModularNeighborhoodPack/Maps/Demo_Map.Demo_Map_C' with ClassGeneratedBy '/Game/ModularNeighborhood

Pack/Maps/Demo_Map.Demo_Map:PersistentLevel.Demo_Map'

LogSlate: Took 0.007557 seconds to synchronously load lazily loaded font '../../../Engine/Content/Editor/Slate/Fonts/DroidSansMono.ttf' (77K)

---- On Thu, 18 May 2017 17:26:10 -0500 lovettchris &lt;notifications@github.com&gt; wrote ----

One more thing, if your pixhawk got armed somehow then it will not report home position and this can also cause DroneShell to hang. If you disarm the drone using radio then DroneShell should be happier.

eanmikale, 2017-05-19:
I also tried to compile the build and was unable to do so. It told me I needed the AirSim Plugin. Here is the error log. 

LogPlatformFile: Not using cached read wrapper

LogInit:Display: Loading recent project file: ../../../../../../Users/Ean/Documents/Unreal Projects/MyProject/MyProject.uproject

LogInit:Display: RandInit(-2145861997) SRandInit(-2145861997).

LogTaskGraph: Started task graph with 5 named threads and 6 total threads with 1 sets of task threads.

LogStats: Stats thread started at 0.059292

LogInit: Using libcurl 7.47.1

LogInit:  - built for x86_64-pc-win32

LogInit:  - supports SSL with OpenSSL/1.0.2g

LogInit:  - supports HTTP deflate (compression) using libz 1.2.8

LogInit:  - other features:

LogInit:      CURL_VERSION_SSL

LogInit:      CURL_VERSION_LIBZ

LogInit:      CURL_VERSION_IPV6

LogInit:      CURL_VERSION_ASYNCHDNS

LogInit:      CURL_VERSION_LARGEFILE

LogInit:      CURL_VERSION_IDN

LogInit:  CurlRequestOptions (configurable via config and command line):

LogInit:  - bVerifyPeer = false  - Libcurl will NOT verify peer certificate

LogInit:  - bUseHttpProxy = false  - Libcurl will NOT use HTTP proxy

LogInit:  - bDontReuseConnections = false  - Libcurl will reuse connections

LogInit:  - CertBundlePath = nullptr  - Libcurl will use whatever was configured at build time.

LogInit: Build: ++UE4+Release-4.15-CL-3416026

LogInit: Engine Version: 4.15.2-3416026+++UE4+Release-4.15

LogInit: Compatible Engine Version: 4.15.0-3299760+++UE4+Release-4.15

LogInit: Net CL: 3299760

LogDevObjectVersion: Number of dev versions registered: 14

LogDevObjectVersion:   Dev-Blueprints (B0D832E4-1F89-4F0D-ACCF-7EB736FD4AA2): 8

LogDevObjectVersion:   Dev-Build (E1C64328-A22C-4D53-A36C-8E866417BD8C): 0

LogDevObjectVersion:   Dev-Core (375EC13C-06E4-48FB-B500-84F0262A717E): 2

LogDevObjectVersion:   Dev-Editor (E4B068ED-F494-42E9-A231-DA0B2E46BB41): 14

LogDevObjectVersion:   Dev-Framework (CFFC743F-43B0-4480-9391-14DF171D2073): 22

LogDevObjectVersion:   Dev-Mobile (B02B49B5-BB20-44E9-A304-32B752E40360): 0

LogDevObjectVersion:   Dev-Networking (A4E4105C-59A1-49B5-A7C5-40C4547EDFEE): 0

LogDevObjectVersion:   Dev-Online (39C831C9-5AE6-47DC-9A44-9C173E1C8E7C): 0

[... 1150 lines cut ...]
LogTemp:Display: Shader map FDebugViewModeMaterialProxy Driveway_Brick_Old_01 complete, GameThreadShaderMap 0x0000021EF6F08F00, marking material FDebugViewModeMaterialProxy Wood_Cladding_Painted_02 as finished

LogTemp:Display: Marking material as finished 0x0000021EF510E500

LogTemp:Display: Shader map FDebugViewModeMaterialProxy Driveway_Brick_Old_01 complete, GameThreadShaderMap 0x0000021EF6F08F00, marking material FDebugViewModeMaterialProxy Brick_01_Dirt as finished

LogTemp:Display: Marking material as finished 0x0000021EF6E9BD00

LogTemp:Display: Shader map FDebugViewModeMaterialProxy Driveway_Brick_Old_01 complete, GameThreadShaderMap 0x0000021EF6F08F00, marking material FDebugViewModeMaterialProxy Wood_Cladding_01 as finished

LogTemp:Display: Marking material as finished 0x0000021EF6E9B100

LogTemp:Display: Shader map FDebugViewModeMaterialProxy Driveway_Brick_Old_01 complete, GameThreadShaderMap 0x0000021EF6F08F00, marking material FDebugViewModeMaterialProxy Shingles_01 as finished

LogTemp:Display: Marking material as finished 0x0000021EF6E9AE00

LogTemp:Display: Finished compile of shader map 0x0000021EF6F0AE80

LogTemp:Display: Shader map FDebugViewModeMaterialProxy Power_Line_01 complete, GameThreadShaderMap 0x0000021EF6F0AE80, marking material FDebugViewModeMaterialProxy Power_Line_01 as finished

LogTemp:Display: Marking material as finished 0x0000021EDFEBEB00

LogTemp:Display: Shader map FDebugViewModeMaterialProxy Power_Line_01 complete, GameThreadShaderMap 0x0000021EF6F0AE80, marking material FDebugViewModeMaterialProxy Power_Line_Cable_01 as finished

LogTemp:Display: Marking material as finished 0x00000

eanmikale, 2017-05-25:
I was able to get the controller working properly. Thank you. I believe I needed to create a separate root environmental variable for AIRSIM, in addition to EIGEN_ROOT.

closed 2017-05-25 by eanmikale

eanmikale, 2017-08-31:
Chris, 

I hope this finds you well. My name is Ean Mikale, and we are currently partnering with the U.S. Department of Labor, and would like to briefly discuss Microsoft's prospective involvement in the apprenticeable occupation of Commercial Drone Pilots/and Commercial Drone Software Developers. We hope you will consider the conversation as we believe your organization has much to gain. AirSim is would be very much involved, which is why I reached out to you in particular. More soon. 

Warm Regards, 

Ean Mikale, J.D.