microsoft/AirSim #221: How do I grab images from all the views simultaneously ?

bypatel, 2017-05-16:
With record button in UE editor, I can save one image type at a time by modifying EPIPCameraType in
ReadPixelsNonBlocking. How do I save them all to be used later for groundtruthing purposes ?

ChaceYoung, 2017-05-18:
Same question here.

saihv, 2017-05-18:
Definitely not the greatest solution, but if you're not too concerned about the frequency of logging, I guess you could just add an argument to ReadPixelsNonBlocking with the EPIPCameraType, and call it three times within Run()?

lukaszdworako, 2017-05-29:
If you are still having issues, the easiest way to "sync" various video feeds would be to create an event that is common between them i.e. a block falling and hitting something would be the first thing that comes to mind since you could sync them up fairly easily based on when the objects hits the ground.

Mayankm96, 2017-06-09:
Any updates @bypatel ?

lovettchris, 2017-06-09:
You can fetch all 3 views using Python (see PythonClient/camera.py).  But there are [performance implications](https://github.com/Microsoft/AirSim/blob/master/docs/camera_views.md#performance).

sytelus, 2017-07-11:
Please see [new APIs](https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md) that allows you to do this!

closed 2017-07-11 by sytelus

sytelus, 2017-12-12:
The feature to record images from multiple cameras simultaneously is now available. You can, for example, record scene and depth images simultaneously. 

Please see the doc: https://github.com/Microsoft/AirSim/blob/master/docs/settings.md#recording