microsoft/AirSim #222: Changing Camera Paramters

Mayankm96, 2017-05-16:
How can I change the position of the camera so that it faces downwards instead of front?

Mayankm96, 2017-05-16:
I found a fix for this by ejecting from the view when the airdrone has been spawned, and selecting the Drone actor. After that I changed the rotation along Y- axis to -90 degrees and the camera started facing downwards. 

I think there must be a C++ script which is rendered to spawn the drone. I am unable to find which one refers to the camera orientation.

Dynode, 2017-05-17:
@Mayankm96 Was it the PIP view you rotated, or the external cam? Anyway, the external cam is created and positioned in ASimModeWorldMultiRotor::setupVehiclesAndCamera, lines 66 - 81 (https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/SimMode/SimModeWorldMultiRotor.cpp)

Getting more control over PIP orientation and resolution in C++ would be nice.

Iirc, the default quadcopter model has two cameras attached to it's front in UE4 (browse the AirSim content folder on lowest part of the screen in UE4, open and look in the 3D view page (forgot the name atm)). I'm unsure if these are relevant or not at this point, and whether or not they are defined in C++. Perhaps @sytelus knows. Also, I've noticed the PIP render target resolutions is defined as 1280x720 in UE4, can be seen & edited if you open, iirc, the Blueprint for SimHUDWidget (again in the content browser -> AirSim folder).

Mayankm96, 2017-05-18:
@Dynode Yeah I was referring to the LeftPipCamera. The way I was doing it seems kind of temporary since I have to reset the orientation everytime. How did you open the AirSim Content in UE4?

Dynode, 2017-05-18:
After compiling & running AirSim in Visual Studio, UE4 pops up with the relevant environment / game mode, then go as shown here:
[image]
Click the Viewport tab to see the drone model and attached cameras (Left- & RightPIPCamera). The cameras have Child Actor Class = BP_PIPCamera

Mayankm96, 2017-05-19:
Whenever I compile the AirSim project I get the dronesheel application opening up. Also in my project content file I only see that of the environment I have imported such as the Mountain Landscape or Neighborhood pack. 

I am unable to find the AirSim content anywhere.

EliaTarasov, 2017-05-19:
Im unable to find the content too. However, you dont want to run AirSim project at all, sincwe it's a plugin. You have to create your own Unreal project with this plugin and run it, if i understand your correctly.

Dynode, 2017-05-19:
Hmm, I followed the instructions in https://github.com/Microsoft/AirSim/blob/master/docs/build.md and https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md

Is assume you've run build.cmd, edited the settings file and right-clicked the .uproject file to "Generate Visual Studio project files". Is AirSim set as the Game Mode for the World you've opene in UE4? (
[image])

EliaTarasov, 2017-05-19:
It's just sets "SimGameMode", but my AirSim code is about 2-3 weeks dated.

Dynode, 2017-05-19:
Wait, you probably also have to enable "Show Plugin Content" in this menu:
[image]

EliaTarasov, 2017-05-19:
Exactly!. Now it can be seen. Thanks!

EliaTarasov, 2017-05-19:
@Mayankm96 how exactly did you find "Drone" Actor?
There is no name of an actor that would exactly matches.

Mayankm96, 2017-05-19:
@Dynode Yeah that helped! Thanks a lot! :)

@EliaTarasov Well in the World Outliner tab at the upper-right corner of the UE4 Editor, type player and you would find an actor named `Player Start`. Then click on it and press `F`. The screen would automatically focus onto the actor.

EliaTarasov, 2017-05-19:
@Mayankm96 thanks for the hint, but this is not what i was looking for.
I need to know either location and orientation of the BodyMesh (Quadrotor) or location and orientation of one of the two of it's cameras. I can get FlyingPawn with this approach:

//Iterate Actors to find AFlyingPawn one. It should be only one though
  for (TActorIterator<AFlyingPawn> Iterator(GetWorld()); Iterator; ++Iterator)
  {
    UAVPawn = dynamic_cast<AFlyingPawn*>(*Iterator);
    if(UAVPawn) {
      // If we find FlyingPawn, lets try to find it's child cameras
      TArray <AActor *> ChildActors;
      UAVPawn->GetAllChildActors(ChildActors);
      if(ChildActors.Num()) {
        //If at least one child exists, iterate all child actors to find PipCameras
        for (int i = 0; i < ChildActors.Num(); ++i)
        {
          AActor *Component = ChildActors[i];
          FString Name = Component->GetName();
          FVector CameraLocation = Component->GetActorLocation();
          FRotator CameraOrientation = Component->GetActorRotation();
        }
      }
    }
  }

This works, i can find both PIP cameras. However, they are not strictly fixed to body, since they locations and orientations are "floating" around the quad. The sort of "Fly-By" views.

Mayankm96, 2017-05-19:
Oh I see. I think you can do that by opening the blueprint of the BP_FlyingPawn (double click on it) as shown by Dynamo and then open the `Viewport` tab . Over there you can edit components or add new meshes. The actor has components in it one of which is the PIP camera. They are defined relative to the origin of the actor. When the actor is spawned into the scene then all of its component stays fixed relative to the actor's origin. 

I was able to change the orientation of the LeftPIPCamera like this and it seems to be working just fine. Let me know if this helps!

[image]

EliaTarasov, 2017-05-19:
Yes! It helps! Again, it was not exactly i was looking for, but you gave me hint why i did mistake with my c++ code. My second camera which was supposedly attached to the quad in fact was not attached. So i left blueprints intact and instead attached camera by calling ``
this->AttachToActor(Cam, FAttachmentTransformRules(EAttachmentRule::SnapToTarget, true));``
And it worked as it was expected.
Thanks for the hint!

Mayankm96, 2017-05-19:
Glad it worked! :)

closed 2017-05-19 by Mayankm96

dasmehdix, 2021-03-02:
> NB: The following regards the 2017.05.16 build of AirSim.
> 
> @Mayankm96 Was it the PIP view you rotated, or the external cam? Anyway, the external cam is created and positioned in ASimModeWorldMultiRotor::setupVehiclesAndCamera(), lines 66 - 81 (https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/SimMode/SimModeWorldMultiRotor.cpp)
> 
> Getting more control over PIP orientation and resolution in C++ would be nice.
> 
> EDIT:
> _There doesn't seem to be any code that specifically sets the camera angle, it's defined in UE4:_
> 
> The BP_FlyingPawn quadcopter Blueprint has two cameras attached to it's front in UE4 (browse the AirSim content folder on lowest part of the screen in UE4, open BP_FlyingPawn and look in the Viewport tab.
> [image]
> )
> **The left one, "LeftPIPCamera" is referred to** in FlyingPawn.cpp (**AFlyingPawn::setupComponentReferences()** )
> **This is the PIP camera. In UE4 you can set orientation and position wrt the drone model.**
> 
> **EDIT:**
> Also; the PIP render target resolutions seem to be defined as 1280x720 (in UE4) in each PIP's RenderTarget found in Air Sim Content -> HUDAssets folder. I'm still not sure if these numbers are overridden elsewhere or actually used.

I had same issue. I changed rotation values of "FrontCamera" object  inside "BP_FlyingPawn" and now camera looks downward which I wanted to set. So, my problem solved. Thanks..