microsoft/AirSim #223: Issue with Hexacopter make

lukaszdworako, 2017-05-17:
Every attempt to execute the make command for the Hexacopter I receive cmake errors:
`CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_ASM_COMPILER not set, after EnableLanguage`
Any advice on how to resolve these errors would be helpful. Tested on multiple linux based environments with same issues throughout

lovettchris, 2017-05-17:
Need more detail, what are you making exactly?  When you say Hexacopter, are you referring to [this page on building a hexacopter](https://github.com/Microsoft/AirSim/wiki/hexacopter)?

lukaszdworako, 2017-05-17:
Sorry, I was referring to that tutorial. I am attempting to run the "make px4fmu-v2_default" after modifying the relevant file.

lovettchris, 2017-05-17:
On Linux or Windows? See [Linux build instructions for PX4 firmware](https://dev.px4.io/en/setup/dev_env_linux.html)

lukaszdworako, 2017-05-19:
CMake build I have is currently incompatible with the PX4 build instructions. Try to reinstall according to PX4 exact instructions

closed 2017-05-19 by lukaszdworako