microsoft/AirSim #226: Load Custom 3D-Model Of A Quad into AirSim

EliaTarasov, 2017-05-22:
Does anybody have an idea or an experience how to load custom 3D Unreal quad model into AirSim?
Within plugin's Content directory there are many .uasset files, but there are no references to these files into the code itself. So i suppose these files are used by Unreal itself. 
However it's not clear how to add custom files to the AirSim to be able to use them.
Thanks!

lovettchris, 2017-05-23:
This page on [building custom hexacopter ](https://github.com/Microsoft/AirSim/wiki/hexacopter) might help give you what you need.

EliaTarasov, 2017-05-23:
Thanks, yesterday i figure out where i should begin and the page you mentined is also very helpful. 
I followed instructions and it worked ok.

closed 2017-05-23 by EliaTarasov