microsoft/AirSim #227: Baro #0 fail: STALE!

Timebutt, 2017-05-23:
Hi,

I'm getting a lot of `Baro #0 fail: STALE!` errors lately, actually every flight I do seems to have one. I don't seem to find anything in the recent commits that might cause this sensor error. Am I missing something, or something else going on?

[image]

[image]

lovettchris, 2017-05-24:
Yeah I've seen that too, try playing with the [barometer noise ](https://github.com/Microsoft/AirSim/commit/f78f11774055fa7381c9cdf6dc08265c93864226)...

sytelus, 2017-05-25:
Did commenting out those line solved the baro STALE issue? May be we should look at PX4 code to see what this message means...

Timebutt, 2017-05-26:
I haven't had the chance to test yet. Will have access to my machine at work earliest on monday, will let you know!

Timebutt, 2017-06-01:
The recent commit fixed the issue at hand!

closed 2017-06-01 by Timebutt