microsoft/AirSim #228: Resolution of views

Thusheep, 2017-05-23:
I want to upscale the segmentation and depth resolution. Where and how can I cange the resolution of the three views in the code? Right now when I record the images I get 1280x720p for the RGB-Image, but only  256x144p for the segmentation and depth-image.

saihv, 2017-05-25:
The resolution for the three views can be changed through Content/HUDAssets/FpvRenderTarget or DepthRenderTarget or SegRenderTarget uasset files. These can be opened in Unreal Engine and you can see the resolution as part of the "Texture Render Target" settings.

Thusheep, 2017-05-26:
Thanks a lot! Do you also know how to change the camera direction of the quadcopter? For example instead looking straight, to look downwards (to get images with bird's eye perspective).

Mayankm96, 2017-05-26:
Already there in the issue #[222](https://github.com/Microsoft/AirSim/issues/222)

Thusheep, 2017-05-26:
Oh perfect. Thank you!

closed 2017-05-26 by Thusheep

Dynode, 2017-05-26:
Do any of you know where to set the main resolution?

Dynode, 2017-05-31:
Main resolution can be set by running
this->PlayerOwner->ConsoleCommand(TEXT("setRes 1920x1080f"));
at start, where "f" denotes "go fullscreen".