microsoft/AirSim #230: Can I Fly Multiple Vehicles at Once

exploke, 2017-05-23:
Is there any way I can fly multiple quads at once in one simulation? If so, how would I proceed to do doing so?

sytelus, 2017-05-25:
Duplicate: https://github.com/Microsoft/AirSim/issues/92

closed 2017-05-25 by sytelus

saihv, 2017-05-25:
@exploke

Before the AirSim code moved to C++ from blueprints, I was able to achieve this by doing three things. But this only works with the older version of AirSim.

1. Replicate the FlyingPawn and PIPCamera assets to represent multiple drones
2. Replicate the Spawn vehicle blueprint within Unreal.
3. Once you ensure multiple vehicles are being spawned, you need to instantiate the multirotor connector within the code n number of times, where n is the number of drones you have.

A basic working implementation for two vehicles can be seen at my fork (https://github.com/saihv/airsim/tree/two_vehicle). This can be extended to more, depending on how much processing power you have.

exploke, 2017-05-25:
Ok, sounds good. Thanks @sytelus and @saihv

spencerh-b, 2017-10-25:
@exploke did this work?

exploke, 2017-10-25:
**@spencerh-b** it did work. AirSim has a dedicated page for that now though. Look at **_docs/multi_vehicle.md_** for more information.

dastoqc, 2017-10-25:
@exploke you manage to have multiple vehicles work as in the docs/multi_vehicle.md ? I had to rollback to a commit in mid-september (https://github.com/Microsoft/AirSim/commit/1fe5e403db32c2adac21a4e920cea75b6118b903) in order to have the possibility to select different controller for each spawned units.

spencerh-b, 2017-10-25:
@exploke I have followed the multi_vehicles.md doc, and I have been able to spawn multiple vehicles, but I have not been able to control these vehicles via an api, do you have any guidance or steps I should take to figure out how do control the vehicle independently?

DavidLSmyth, 2017-12-04:
>  I have not been able to control these vehicles via an api, do you have any guidance or steps I should take to figure out how do control the vehicle independently?

I'd also appreciate advice on this, I have multiple drones in the environment but am not sure how to control each of them individually via an api.

dastoqc, 2017-12-04:
@spencerh-b and @DavidLSmyth, if you start from the commit I pointed above, then you can edit the setting.json as explained here: https://github.com/Microsoft/AirSim/blob/master/docs/multi_vehicle.md and use different ports. When you create an instance in Unreal just set the controller name to whatever is defined in your settings.json.

DavidLSmyth, 2017-12-05:
Thanks for the reply @dastoqc. I have two BP_FlyingPawns dropped into the environment. My settings.json looks like this:
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.0,

  "SimpleFlight1": {
  	"FirmwareName": "SimpleFlight",
    "ApiServerPort": 41451
  },
  "SimpleFlight2": {
  	"FirmwareName": "SimpleFlight",
    "ApiServerPort": 41452
  }
}

The python API code can connect and arm the drone etc. via port 41451 but when I try and run the same code on a separate process on port 41452 I get a ConnectionRefusedError when trying to connect. I'm certain what you mean by 

> just set the controller name

is this meant to be in the VehicleConfig section? I can't seem to find the VehicleConfig section in the details of the BP_FlyingPawn that I dropped into the environment which seems to be the issue. Under details I can see Transform, Debugging, Pawn, Camera, Rendering, Input and actor sections. Apologies if I have missed anything obvious.

dastoqc, 2017-12-05:
@DavidLSmyth which version of Airsim plugin are you using? The Unreal blueprint methods were re-written a couple of months ago and the 'VehicleConfig' is not available anymore. This is why I stick with commit 1fe5e40.

DavidLSmyth, 2017-12-05:
@dastoqc I'm running the most up to date version. Thanks very much for the info - I'll clone that commit and follow your instructions. If would be great to get the functionality to work with the most up to date version but that will be a good starting point

DavidLSmyth, 2017-12-05:
I have the blocks example from commit 1fe5e40 running with two drones with Vehicle Config Names SimpleFlight_1 and SimpleFlight_2 and my settings.json looks like: 

{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingdVersion": 1.0,

  "SimpleFlight_1": {
    "FirmwareName": "SimpleFlight",
    "DefaultVehicleState": "Armed",
    "ApiServerPort": 41453
  },
  "SimpleFlight_2": {
    "FirmwareName": "SimpleFlight",
    "DefaultVehicleState": "Armed",
    "ApiServerPort": 41451
  }
}

but I'm still getting a connection refused error for port 41451. Have also tried with more vehicles - only the first listed in settings.json can connect. Any further advice would be greatly appreciated.

dastoqc, 2017-12-05:
don't know if it's related but you seem to have a typo. This settings.json works just fine here:
`{
  "SettingsVersion": 1.0,
  "DefaultVehicleConfig": "SimpleFlight",

  "SimpleFlight": {
    "FirmwareName": "SimpleFlight",
    "DefaultVehicleState": "Disarmed",
    "ApiServerPort": 41451
  },
  "SimpleFlight_1": {
    "FirmwareName": "SimpleFlight",
    "DefaultVehicleState": "Disarmed",
    "ApiServerPort": 41452
  },
  "SimpleFlight_2": {
    "FirmwareName": "SimpleFlight",
    "DefaultVehicleState": "Disarmed",
    "ApiServerPort": 41453
  }
}`

I used the HelloDrone example and duplicate the rpcclient commands for other ports.

Hope it helps.

DavidLSmyth, 2017-12-05:
The typo actually is a bug, it's written when settings.json gets created. I've copied your settings.json exactly (apart from the typo, which if I correct I get a warning in the unreal viewport) but can only connect to the drone that has Is Fpv Vehicle checked .Was there any additional setup in Unreal regarding cameras or anything like that? 

Think I may have to try and output some debugging messages from the c++ code, not sure where else to go from here.  To test I'm connecting directly to a tcp socket on localhost and each port and sending bytes using messagePack encoding over tcp, assuming that's not the problem since it works for a single drone:

```python
import socket
import sys
port = sys.argv[1]
local_sock = socket.socket()
local_sock.connect(('127.0.0.1', int(port)))
print(local_sock.getsockname())

local_sock.sendall(b'\x94\x00\x01\xb0enableApiControl\x91\xc3')
data = local_sock.recv(1024)
for i in data:
    print(i)
print('arming')
local_sock.sendall(b'\x94\x00\x02\xa9armDisarm\x91\xc3')
data = local_sock.recv(1024)
for i in data:
    print(i)
print('printing to viewport')
#local_sock.sendall(b'\x94\x00\x03\xb2simPrintLogMessage\x93\xaaTaking Off\xa0\x00')
local_sock.send(b'\x94\x00\x04\xa7takeoff\x91\x05')
```

 Thanks very much for the continued help @dastoqc

SaiVineethJ, 2018-02-20:
Try the codes from 
https://github.com/saihv/AirSimMulti
 and change the settings.json as:
{
  "SettingsVersion": 1.0,
  "DefaultVehicleConfig": "SimpleFlight",

  "SimpleFlight": {
    "FirmwareName": "SimpleFlight",
    "DefaultVehicleState": "Disarmed",
    "ApiServerPort": 41451
  },
  "SimpleFlight_1": {
    "FirmwareName": "SimpleFlight",
    "DefaultVehicleState": "Disarmed",
    "ApiServerPort": 41452
  },
  "SimpleFlight_2": {
    "FirmwareName": "SimpleFlight",
    "DefaultVehicleState": "Disarmed",
    "ApiServerPort": 41453
  }
}
This works fine for 3 drones.