microsoft/AirSim #231: HITL, "Could not open the serial port" "error=5"

Dynode, 2017-05-23:
Windows 10. Running HITL with Pixhawk PX4 and QGroundControl. QGC sees the device. Windows Device mgr shows the Pixhawk on COM3, looks good. Arming works in QGC.

Upon starting AirSim I get the following message shown in the attached pic:
[image]

Someone recognize the error?

Thanks. :')

lovettchris, 2017-05-24:
"and QGroundControl" ? Perhaps QGroundControl has locked the serial port.  Try closing QGroundControl.  Run AirSim, click Play, let it connect, then you can launch QGroundControl and it will connect via the UDP port published by AirSim.  In otherwords, AirSim can also act as a proxy from Pixhawk to QGroundControl.

Dynode, 2017-05-24:
Ah, of course. That was it. Launched QGC last, works now.

Thanks.

closed 2017-05-24 by Dynode