microsoft/AirSim #232: How can I get the 2D location and 3D location of objects in the scene? For example, when the drone

Geekking, 2017-05-24:


sytelus, 2017-05-25:
Yes, you can but you will need to dig in to Unreal Engine programming. Basically environment has all these meshes so you can iterate through meshes that interests you and then there are direct methods to get its position and size in world coordinates. If you have never done any Unreal engine programming, its non-trivial, however.

closed 2017-05-25 by sytelus

Geekking, 2017-05-26:
OK, thanks.

cangokalp, 2017-10-25:
@Geekking Have you figured this out? I wish there was an easy way to get the coordinates and bounding box of the objects in the environment too. I need the same thing

alessandro-montanari, 2017-11-02:
@Geekking looking for the same thing here.

evroon, 2021-03-07:
Probably way too late, but for future reference: You can get bounding boxes without modifying the source code using the segmentation outputs.

First, set the object IDs of everything to zero, except the drone (if you want to get bounding boxes around the drone):
```python
client.simSetSegmentationObjectID("[\w]*", 0, True)
client.simSetSegmentationObjectID("Drone[\w]*", 255, True)
```
Save the segmentation images as in the [hello-drone example](https://microsoft.github.io/AirSim/apis/#hello-drone).

Now you can calculate the bounding box simply by checking which rows and columns contain non-zero pixels: (there is probably a more efficient way, but it works)

```python
import numpy as np
import cv2

img = cv2.imread('path_to_image.png')

height, width = img.shape[:2]
img_size = np.array([width, height])
start_x, start_y = -1, -1
end_x, end_y = -1, -1

for y in range(height):
    if np.sum(img[y, :, :]) > 0.0:
        end_y = y

        if start_y == -1:
            start_y = y

for x in range(width):
    if np.sum(img[:, x, :]) > 0.0:
        end_x = x

        if start_x == -1:
            start_x = x

img = cv2.rectangle(img, (start_x, start_y), (end_x, end_y), (255, 0, 0))
cv2.imwrite('frame.png', img)
```

The (very thin, sorry) blue box shows the bounding box in the acquired segmentation image:
[image]

rancheng, 2021-05-28:
> Probably way too late, but for future reference: You can get bounding boxes without modifying the source code using the segmentation outputs.
>
> First, set the object IDs of everything to zero, except [...]

This will only work when there is no overlapping of the object, if anything obscured this object, 2D bounding box will be off.

evroon, 2021-05-28:
@rancheng Yes that is true, that is exactly what someone noted [here](https://github.com/microsoft/AirSim/pull/3472#issuecomment-807088803) as well. So if you want bounding box detections where occlusion is taking place, you can use PR #3472.

zohaibjan, 2022-08-09:
@evroon that is exactly what I am trying to do but sort of struggling with the PR #3472. Is there a way to determine if objects are overlapped or occluded ?.

Thanks.

evroon, 2022-08-09:
@zohaibjan what do you mean exactly? Segmentation masks typically don't support occlusion, i.e. multiple segmentation IDs per pixel from what I know.

I don't have experience with that PR you link btw.

zohaibjan, 2022-08-10:
@evroon I am trying to generate object detection database but so far cannot isolate occluded objects from ground truth. I have tried multiple openCV "tricks" but there is still a level of stochasticity in it left which could be detrimental for the training algorithm. So need more deterministic approach rather than a random one.

evroon, 2022-08-11:
This solution should be fully deterministic, every pixel belonging to the object you want to segment should be highlighted in the mask. I'm not really sure what your exact problem is.

zohaibjan, 2022-08-11:
@evroon  think if a car is occluded behind a tree and only half of it is visible. In that case only half of the segmentation is visible even after switching everything else to -1 or 0 in setSegmentationObjectID. So basically, the question is can I determine if the object is occluded using the segmentation

evroon, 2022-08-11:
Ah okay, I don't think that's possible AFAIK, but I haven't done anything with AirSim in a while

zohaibjan, 2022-08-11:
Thanks for the reply though. I am struggling at the moment neither the Object Detection API nor segmentation AFAIK can support this.

Vanessa-Yehui, 2023-10-24:
I wonder if we get the world coordinates, depth image and camera parameters, if projection possible or segmentation better?