microsoft/AirSim #233: HItL, default quadrotor (BP_FlyingPawn) is unstable

Dynode, 2017-05-24:
I experience unstable behavior with the default drone model even in a clean AirSim build. I suspect I'm missing something critical about how AirSim works with the Pixhawk -- does the latter need to be tuned in Px4, or should AirSim take care of that? AirSim simulates sensor data?

I've attached a screenshot with sensor data on the right:
[image]

Upon taking of, the drone begins to oscillate violently or drift and crash (depending on weight params set in Px4MultiRotor::setupFrameGenericQuad). Any idea what the source of this is?

Dynode, 2017-05-25:
Update: When the physical constants for the drone model is ok, it seems this is an issue only when the RC is not connected, like when running the "takeoff" command in DroneShell. When the constants are way off, the drone get unstable. That's my thought anyways.

sytelus, 2017-05-25:
Which physical constants? Rotors or drones? Could you please post example values that produces bad behavior?

Dynode, 2017-05-25:
Alright, I edited the previous post, can't reset the to the bad consts atm, will do tomorrow but iirc, bad oscillations happened in manual flight (RC connected) with default RotorParams values and too small motor_assembly_weight in Px4MultiRotor.hpp::setupFrameGenericQuad() after adjusting arm length to about 70 cm. I'm sorry I can't be more specific atm but I'm pretty sure at this time that some of my troubles are due to bad calibration and just plain wrong params. :')

sytelus, 2017-08-14:
Closing this as this was issue to specific setup.

closed 2017-08-14 by sytelus