microsoft/AirSim #234: Custom quadrotor model collision box problem, won't stand upright

Dynode, 2017-05-24:
I've imported a custom model (Gryphon Dyn. X8, long legs), the rotors are set accordingly as per instructions / the default quadrotor BP_FlyingPawn. The frame and rotors are a bit larger than the default model / pawn. Model origo is such that the rotors are about the same height / offset as on the default setup in UE4 (I have yet to understand line 57 - 59 in Px4MultiRotor.cpp, "set up dimensions of core body box or abdomen (not including arms)").

The models just plain refuses to stand upright; it slowly tips over and slowly spins randomly while the collision count is constantly increasing. There is clearly something going on with the physics sim part on this model -- is there anything that has to be done in UE4, or does AirSim detect model boundaries?

Thanks again.

lovettchris, 2017-05-25:
> The models just plain refuses to stand upright; it slowly tips over and slowly spins randomly while the collision count constantly increases

I had the same problem with my hexacopter model, something to do with the long legs, and ground contract not working properly.  This is a physics bug of some sort, we are looking into it...   In the meantime you should be able to take off before this starts happening and fly ok, at least that's what I could do with my hexacopter model.

Dynode, 2017-05-25:
@lovettchris Thanks, I'll try to look into it as well, compare our two models.

sytelus, 2017-05-25:
I think there is a problem with ground detection code... Right now I'm travelling but hope to experiment with potential fix around 15th June after I get back...

Dynode, 2017-05-29:
Solved this to the point where starting AirSim and resetting the drone is stable in almost all cases. Now what happens is the height at which the model comes to rest at on the ground varies a little.

closed 2017-05-29 by Dynode