microsoft/AirSim #235: Drone Shell GPS Lock

eanmikale, 2017-05-25:
Having issues getting the Drone Shell to gain a GPS lock. I've made sure that QGround Control is not open. The simulator is running, and when I execute the DroneShell and wait for the GPS to lock, it will not do so no matter how long I wait. Any ideas on troubleshooting? Thanks again.

lovettchris, 2017-06-02:
Are you using HITL mode with Pixhawk hardware or SITL mode ?  If HIL mode did you select the "HIL Quadrocopter X" AirFrame using QGroundControl?

eanmikale, 2017-06-04:
Yes. I am using HITL mode with Pixhawk hardware (IRIS+). In HIL I have selected the "HIL Quadcopter X" Airframe using QGroundControl. In QGroundControl I still do not see a GPS Lock. When first close QGroundControl, and then open up Unreal Engine, and open up the landscape mountains project, or the Modular Neighborhood project, press play, and then go back into the AirSim folder and select DroneShell, it continues to read "Waitinf for drone to report a valid GPS location." I wonder if two functions are utilizing the same port, not allowing the GPS to be read by the either QGroundControl or DroneShell. Hopefully, you have some thoughts on this. Once we get past this challenge, we'll be home free as far as what we need. Thank you for your hard work again.

eanmikale, 2017-06-04:
I also read somewhere that it may be the local proxy? If internet connect has anything to do with it, our office internet has some hoops to jump through. Just a thought.

clovett, 2017-06-05:
Sounds like you are doing everything correctly.  It should work.  It would help if you could provide some logs, as per [logging](https://github.com/Microsoft/AirSim/blob/master/docs/logging.md).  You can also look at the hilgps messages being sent by AirSim using [LogViewer](https://github.com/Microsoft/AirSim/blob/master/docs/log_viewer.md).

eanmikale, 2017-06-07:
I found the issue. One of the port numbers was off by 1. Everything works beautifully. Thank you.

closed 2017-06-07 by eanmikale

lovettchris, 2017-06-07:
great!

eanmikale, 2019-02-08:
Chris,
We'd like to discuss a potential partnership in the education space as we release a commercial product to place nice with AirSim. Please let us know how to connect. More soon. 

Warm Regards,

Ean Mikale, J.D.

Chief Innovation Officer

Infinite 8 Institute, L3C

ean@infinite8institute.com

www.infinite8institute.com

U.S. Federal Contractor

---- On Fri, 02 Jun 2017 16:40:41 -0500 notifications@github.com wrote ----

Are you using HITL mode with Pixhawk hardware or SITL mode ? If HIL mode did you select the "HIL Quadrocopter X" AirFrame using QGroundControl?