microsoft/AirSim #238: Lighting effect issue (level is Neighboor_Modular)

Dynode, 2017-05-26:
I'm sorry if this turns out to be irrelevant to AirSim but I post it here in case it is, it might be my own fault from having modified some internals in the PIPCamera and CameraDirector classes.

There is a weird lighting effect affecting the PIP Scene view. On ground, and up until a certain height, it blackens out everything in the PIP. From certain height, the effect fades out but is clearly defined to a square around the player.

Screenshots with better descriptions are here: http://imgur.com/a/d8PwC 
(In case you wonder why the fog on the level is gone, it's just disabled.)

Can this just be me hooking up things wrong wrt camera or views? Is there a shader causing this? Any help is appreciated.

lovettchris, 2017-05-27:
I agree there's some sort of bug here, we need to investigate.

Dynode, 2017-05-27:
A temporary workaround can be to automatically call the event that's triggered when selecting the pawn / drone in Play-in-Editor ("ejected" view, not standalone game). Perhaps it's possible to see what methods are called in some sort of debug view, and then just call them in the setupVehiclesAndCamera() method.

Dynode, 2017-05-27:
Actually, the bug disappeared when I scaled the pawn back to 1.0 instead of 0.1 (I compensated for this by scaling the imported drone model instead). Cameras placed in the pawn were also scaled back accordingly (in the Viewport tab). Unexpected behavior that scale should affect whatever lighting effect is at play.

closed 2017-05-29 by Dynode