microsoft/AirSim #241: HiTL, After crashing in AirSim, drone immediately wants to tilt and fly left, collision seems to bug out

Dynode, 2017-05-29:
Landing and starting seems stable, flying is alright except for overly sensitive yaw (seems to be a constant rotation at times). However, if the drone crashes, re-arming and restarting AirSim ( this->PlayerOwner->ConsoleCommand(TEXT("RestartLevel")); )  does not stop the drone from aggressively going left during takeoff.

Re-starting the Pixhawk (out & in with the USB cable) works but why is that the only helpful thing to do? Running the latest beta firmware, had the same issue before updating as well.

lovettchris, 2017-05-30:
Yes what you are seeing there is the px4 position estimation gets confused by the crash.  I haven't found a good what to reset that over mavlink, but if you run the logviewer you will see the orientation reported by px4, and you will see it slowly drift back to level, once it does that takeoff will be ok.  What we should do is put an attitude widget on the airsim screen so you can always monitor that, then things will make more sense.

Dynode, 2017-05-31:
I see. An attitude widget would be fine but the issue seems to have resolved itself after I updated to the 20170529 version of AirSim and the latest beta of Px4. That, combined with new legs (not thin T tubes) for our 3D model seems to have fixed most instability issues wrt landing and takeoff.

closed 2017-05-31 by Dynode