microsoft/AirSim #244: How to record depth and Fpv views simultaneously?

Mayankm96, 2017-05-30:


lovettchris, 2017-06-02:
This will be fixed with this [pull request](https://github.com/Microsoft/AirSim/pull/255).

Mayankm96, 2017-06-03:
Well I moved to the new branch and tried running the following script: 
```
# use open cv to show new images from AirSim 

from PythonClient import *
import cv2
import time
import sys

client = AirSimClient('127.0.0.1')

help_d = False
help_s = False

fontFace = cv2.FONT_HERSHEY_SIMPLEX
fontScale = 0.5
thickness = 2
textSize, baseline = cv2.getTextSize("FPS", fontFace, fontScale, thickness)
print (textSize)
textOrg = (10, 10 + textSize[1])
frameCount = 0
startTime=time.clock()
fps = 0

while True:
    # because this method returns std::vector<uint8>, msgpack decides to encode it as a string unfortunately.
    result_d = client.getImageForCamera(0, AirSimImageType.Depth)
 
    if (result_d == "\0"):
        if (not help_d):
            help_d = True
            print("Please press '1' in the AirSim view to enable the Depth camera view")
    else:
        rawImage = np.fromstring(result_d, np.int8)
        png = cv2.imdecode(rawImage, cv2.IMREAD_UNCHANGED)
        
        cv2.putText(png,'FPS ' + str(fps),textOrg, fontFace, fontScale,(255,0,255),thickness)
        cv2.imshow("Depth", png)

    result_s = client.getImageForCamera(0, AirSimImageType.Scene)
  
    if (result_s == "\0"):
        if (not help_s):
            help_s = True
            print("Please press '3' in the AirSim view to enable the Scene camera view")
    else:
        rawImage = np.fromstring(result_s, np.int8)
        png = cv2.imdecode(rawImage, cv2.IMREAD_UNCHANGED)
        
        cv2.putText(png,'FPS ' + str(fps),textOrg, fontFace, fontScale,(255,0,255),thickness)
        cv2.imshow("Scene", png)

    frameCount  = frameCount  + 1
    endTime=time.clock()
    diff = endTime - startTime
    if (diff > 1):
        fps = frameCount
        frameCount = 0
        startTime = endTime
    
    key = cv2.waitKey(1) & 0xFF;
    if (key == 27 or key == ord('q') or key == ord('x')):
        break;
```

However, although the scenes are on, the variables `result_s` and `result_d` are not receiving any string.
```
C:\Users\xam\Desktop\AirSim\PythonClient>python camera.py
(32, 12)
Please press '1' in the AirSim view to enable the Depth camera view
Please press '3' in the AirSim view to enable the Scene camera view
```

Mayankm96, 2017-06-05:
After the new pull request, the `Airsim` Plugin is not running with the Unreal project. The following errors are being shown when I run the simulator:

+		_Ln	0x00000052126cb448 {_MyTarget=unique_ptr (???, ???) }	std::_LaunchPad<std::unique_ptr<std::tuple<void (__cdecl msr::airlib::WorkerThread::*)(void),msr::airlib::WorkerThread *>,std::default_delete<std::tuple<void (__cdecl msr::airlib::WorkerThread::*)(void),msr::airlib::WorkerThread *> > > > *
+		_Ln->_MyTarget	unique_ptr (???, ???)	std::unique_ptr<std::tuple<void (__cdecl msr::airlib::WorkerThread::*)(void),msr::airlib::WorkerThread *>,std::default_delete<std::tuple<void (__cdecl msr::airlib::WorkerThread::*)(void),msr::airlib::WorkerThread *> > >

Mayankm96, 2017-06-07:
THe recording thread with two images being recorded at a time shows a rate of FPS 2. Is there a way to boost the frame rate?

Mayankm96, 2017-06-09:
Also while recording I am seeing that the synchronization between the captured views is not there. @lovettchris

lovettchris, 2017-06-09:
I put some performance info here: [camera_views.md](https://github.com/Microsoft/AirSim/blob/VehicleCamera/docs/camera_views.md), but in terms of synchronization, they will never be perfectly synchronized because they are separate render requests, but we could grab a timestamp as close as possible to that render request and return that too. Would that help?

saihv, 2017-06-09:
If you're constrained by hardware, perhaps you could just screen-grab the images? especially if you need near-simultaneous recording. I recently came up with a [little script](https://github.com/saihv/AirSimMulti/blob/swarm/Scripts/cameraLogger.py) that uses Windows and AirSim APIs to record screen captures of all three views from their individual windows as well as ground truth data, perhaps that would work for you too.

lovettchris, 2017-06-10:
nice!

Mayankm96, 2017-06-10:
@lovettchris I am not sure, because the FPS drops quite a bit and even if timestamps are there I don't think match would be possible. One alternative I have is that I fly the drone super-slow so that the difference between two rendering images isn't different. I don't like the idea of this because the drone would be traversing the entire map.

Mayankm96, 2017-06-10:
@saihv I tried your method too. i am just concerned if I start cropping the images then how will the quality be effected. Otherwise I think it's an alternate I might be interested to look into. Thanks!

ShukuiZhang, 2017-06-20:
What @Mayankm96 concerned is also mine. It seems cropped image is too small for my case. Have you come up with other ideas? On the other hand, if @saihv 's idea is the best for now, how can we enlarge the box for depth and scene images?

Alternatively, I've been thinking if we can work directly on the code that generates scene and depth images showed up in the box because it seems they are well synchronized at this end. @lovettchris 

Thanks all!

saihv, 2017-06-20:
@ShukuiZhang Just an FYI, my method doesn't record the image sub views from the game. If you double click on HUDAssets/FpvRenderTarget.uasset (and other render targets), Unreal opens them as a separate window, and that's what is recorded.

ShukuiZhang, 2017-06-20:
Thanks for your reply. @saihv  Applying your method gives me the following image in my case, which is why I mentioned "crop". I assume this is not what to be expected because both saving the image for scene and depth generate these same type of images.
[image]

saihv, 2017-06-20:
That's still the main game window. You'd have to manually double click on
each render target asset so each view is in its own separate window.. and
then adjust the dimensions in the script to get the full image based on
your monitor dimensions.

ShukuiZhang, 2017-06-20:
Thanks so much! @saihv That make more sense to me now. One minor problem is that you mentioned "double clicking" on the uasset files should automatically generate seperate windows for them. But that is not the case at my end.  I assume the uasset files you mentioned is in "Unreal Projects\Myproject\Plugins\AirSim\Content\HUDAssets". When I "double clicking" on the uasset files, I need to choose the application to open them, do you mean to open them in UE4 or something else? But choosing UE4 seems to be weird...  Forgive me if this is so basic.

saihv, 2017-06-20:
Ah, I'm sorry! I meant double click them through Unreal engine. You'd have
to select "show plugin content" in the bottom of the content browser, then
go to AirSim content/HUDAssets.

ShukuiZhang, 2017-06-20:
Yeah, I just found them. Thank you so much for help! @saihv They looks good now.

Mayankm96, 2017-06-27:
@lovettchris I was thinking a bit on how we can record all the views together. One possible idea that sounds reasonable to me is that, we render a 7 channel matrix with 3 channels for Fpv view, 3 channels for segmentation view and 1 channel for depth image. After that we record this single matrix and store it into a xml or yaml file, which we then read using another script where we split the 7 channels into their respective images. 

I am not an expert in UE4 Editor, so I looking for your suggestion on feasibility of this idea.

Mayankm96, 2017-06-27:
Another approach which I am thinking now is to have the three views concatenated into a bigger single image, which we can record and then break into it's constituent images.

sytelus, 2017-07-09:
I have added the new API [simGetImages()](https://github.com/Microsoft/AirSim/blob/master/docs/apis.md) that allows you to get left, right and depth images simultaneously. The API return value is a struct that contains camera position, orientation and time stamp. Please see HelloDrone code for simple example. Also [this code](https://github.com/Microsoft/AirSim/blob/master/Examples/StereoImageGenerator.hpp) has more detailed example.

ShukuiZhang, 2017-07-11:
The synchronization issue between images is still there, right? @sytelus Because the getImage() function is called iteratively in simGetImages() to get all images. Do you have any plan to fix the synchronization issue recently? Thanks.

sytelus, 2017-07-17:
I think synchronization issue should be solved. Even though simGetImages() gets images iteratively, it happens in same tick. Images will have different timestamp but they are generated in same tick.

closed 2017-07-17 by sytelus

ShukuiZhang, 2017-07-19:
Hi @sytelus . I still observed the discrepancy between scene and depth images. The following is what I get from the latest code, both with left camera. 

Scene image:
[image]
Depth image:
[image]
Any comments on what happened? Thanks.