microsoft/AirSim #245: Unable to use MoveToPosition in DroneShell

Mayankm96, 2017-05-30:
I changed the left pip camera orientation to face downwards. However now I need to guide the drone through the environment by giving it way-points. I tried running the HelloDrone and noticed that the `MoveToPosition` leads to asking me to abort the program.  So I tried running the command on DroneShell and and I receive the following error:
```
Async RPC Error: rpclib: function 'moveToPosition' (called with 8 arg(s)) threw an exception. The exception contained this information: lookahead error is continually increasing so we do not have safe control, aborting moveOnPath operation.
```
I believe changing the camera orientation has led to this error. Any idea on how I may correct this?

lovettchris, 2017-05-30:
Pip camera orientation should have nothing to do with moveOnPath behavior.  I made some fixes to HelloDrone, please try again.

Mayankm96, 2017-05-30:
The build is failing now. 
```
Done Building Project "C:\Users\xam\Desktop\AirSim\AirLib\AirLib.vcxproj" (default targets).

InitializeBuildStatus:
  Touching "C:\Users\xam\Desktop\AirSim\DroneShell\temp\DroneShell\x64\Debug\DroneShell.tlog\unsuccessfulbuild".
ClCompile:
  C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin\amd64\CL.exe /c /IC:\Users\xam\Desktop\AirSim\DroneShell\.
  .\AirLib\deps\rpclib\include /Iinclude /IC:\Users\xam\Desktop\AirSim\eigen\Eigen3 /IC:\Users\xam\Desktop\AirSim\Drone
  Shell\..\AirLib\include /ZI /nologo /W4 /WX /Od /D _SCL_SECURE_NO_WARNINGS /D _CRT_SECURE_NO_WARNINGS /D _DEBUG /D _C
  ONSOLE /D _UNICODE /D UNICODE /Gm /EHsc /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"C:\Users\x
  am\Desktop\AirSim\DroneShell\temp\DroneShell\x64\Debug\\" /Fd"C:\Users\xam\Desktop\AirSim\DroneShell\temp\DroneShell\
  x64\Debug\vc140.pdb" /Gd /TP /wd4100 /wd4505 /wd4820 /wd4464 /wd4514 /wd4710 /wd4571 /errorReport:queue /w34263 /w342
  66 src\main.cpp
  main.cpp
c:\users\xam\desktop\airsim\airlib\include\controllers\waiter.hpp(18): fatal error C1073: Internal error involving incr
emental compilation(compiler file 'f:\dd\vctools\compiler\cxxfe\sl\p1\c\esu.c', line 283) [C:\Users\xam\Desktop\AirSim\
DroneShell\DroneShell.vcxproj]
Done Building Project "C:\Users\xam\Desktop\AirSim\DroneShell\DroneShell.vcxproj" (default targets) -- FAILED.

Project "C:\Users\xam\Desktop\AirSim\AirSim.sln" (1) is building "C:\Users\xam\Desktop\AirSim\HelloDrone\HelloDrone.vcx
proj" (5) on node 1 (default targets).
InitializeBuildStatus:
  Creating "C:\Users\xam\Desktop\AirSim\HelloDrone\temp\HelloDrone\x64\Debug\HelloDrone.tlog\unsuccessfulbuild" because
   "AlwaysCreate" was specified.
ClCompile:
  All outputs are up-to-date.
Link:
  All outputs are up-to-date.
  HelloDrone.vcxproj -> C:\Users\xam\Desktop\AirSim\HelloDrone\build\x64\Debug\HelloDrone.exe
  HelloDrone.vcxproj -> C:\Users\xam\Desktop\AirSim\HelloDrone\build\x64\Debug\HelloDrone.pdb (Full PDB)
FinalizeBuildStatus:
  Deleting file "C:\Users\xam\Desktop\AirSim\HelloDrone\temp\HelloDrone\x64\Debug\HelloDrone.tlog\unsuccessfulbuild".
  Touching "C:\Users\xam\Desktop\AirSim\HelloDrone\temp\HelloDrone\x64\Debug\HelloDrone.tlog\HelloDrone.lastbuildstate"
  .
Done Building Project "C:\Users\xam\Desktop\AirSim\HelloDrone\HelloDrone.vcxproj" (default targets).

Project "C:\Users\xam\Desktop\AirSim\AirSim.sln" (1) is building "C:\Users\xam\Desktop\AirSim\DroneServer\DroneServer.v
cxproj" (6) on node 1 (default targets).
InitializeBuildStatus:
  Touching "C:\Users\xam\Desktop\AirSim\DroneServer\temp\DroneServer\x64\Debug\DroneServer.tlog\unsuccessfulbuild".
ClCompile:
  C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin\amd64\CL.exe /c /IC:\Users\xam\Desktop\AirSim\DroneServer\
  ..\AirLib\deps\rpclib\include /Iinclude /IC:\Users\xam\Desktop\AirSim\eigen\Eigen3 /IC:\Users\xam\Desktop\AirSim\Dron
  eServer\..\MavLinkCom\include /IC:\Users\xam\Desktop\AirSim\DroneServer\..\AirLib\include /ZI /nologo /W4 /WX /Od /D
  _SCL_SECURE_NO_WARNINGS /D _CRT_SECURE_NO_WARNINGS /D _DEBUG /D _CONSOLE /D _UNICODE /D UNICODE /Gm /EHsc /RTC1 /MDd
  /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"C:\Users\xam\Desktop\AirSim\DroneServer\temp\DroneServer\x64\
  Debug\\" /Fd"C:\Users\xam\Desktop\AirSim\DroneServer\temp\DroneServer\x64\Debug\vc140.pdb" /Gd /TP /wd4100 /wd4505 /w
  d4820 /wd4464 /wd4514 /wd4710 /wd4571 /errorReport:queue /w34263 /w34266 main.cpp
  main.cpp
c:\users\xam\desktop\airsim\airlib\include\controllers\waiter.hpp(18): fatal error C1073: Internal error involving incr
emental compilation(compiler file 'f:\dd\vctools\compiler\cxxfe\sl\p1\c\esu.c', line 283) [C:\Users\xam\Desktop\AirSim\
DroneServer\DroneServer.vcxproj]
Done Building Project "C:\Users\xam\Desktop\AirSim\DroneServer\DroneServer.vcxproj" (default targets) -- FAILED.

Project "C:\Users\xam\Desktop\AirSim\AirSim.sln" (1) is building "C:\Users\xam\Desktop\AirSim\MavLinkCom\MavLinkTest\Ma
vLinkTest.vcxproj" (7) on node 1 (default targets).
InitializeBuildStatus:
  Creating "C:\Users\xam\Desktop\AirSim\MavLinkCom\MavLinkTest\temp\MavLinkTest\x64\Debug\MavLinkTest.tlog\unsuccessful
  build" because "AlwaysCreate" was specified.
ClCompile:
  All outputs are up-to-date.
Link:
  All outputs are up-to-date.
  MavLinkTest.vcxproj -> C:\Users\xam\Desktop\AirSim\MavLinkCom\MavLinkTest\build\x64\Debug\MavLinkTest.exe
  MavLinkTest.vcxproj -> C:\Users\xam\Desktop\AirSim\MavLinkCom\MavLinkTest\build\x64\Debug\MavLinkTest.pdb (Full PDB)
BscMake:
  All outputs are up-to-date.
FinalizeBuildStatus:
  Deleting file "C:\Users\xam\Desktop\AirSim\MavLinkCom\MavLinkTest\temp\MavLinkTest\x64\Debug\MavLinkTest.tlog\unsucce
[... 40 lines cut ...]

"C:\Users\xam\Desktop\AirSim\AirSim.sln" (default target) (1) ->
"C:\Users\xam\Desktop\AirSim\DroneShell\DroneShell.vcxproj" (default target) (2) ->
(ClCompile target) ->
  c:\users\xam\desktop\airsim\airlib\include\controllers\waiter.hpp(18): fatal error C1073: Internal error involving in
cremental compilation(compiler file 'f:\dd\vctools\compiler\cxxfe\sl\p1\c\esu.c', line 283) [C:\Users\xam\Desktop\AirSi
m\DroneShell\DroneShell.vcxproj]

"C:\Users\xam\Desktop\AirSim\AirSim.sln" (default target) (1) ->
"C:\Users\xam\Desktop\AirSim\DroneServer\DroneServer.vcxproj" (default target) (6) ->
  c:\users\xam\desktop\airsim\airlib\include\controllers\waiter.hpp(18): fatal error C1073: Internal error involving in
[... cut ...]
"C:\Users\xam\Desktop\AirSim\AirSim.sln" (default target) (1) ->
"C:\Users\xam\Desktop\AirSim\AirLibUnitTests\AirLibUnitTests.vcxproj" (default target) (9) ->
  c:\users\xam\desktop\airsim\airlib\include\controllers\waiter.hpp(18): fatal error C1073: Internal error involving in
cremental compilation(compiler file 'f:\dd\vctools\compiler\cxxfe\sl\p1\c\esu.c', line 283) [C:\Users\xam\Desktop\AirSi
m\AirLibUnitTests\AirLibUnitTests.vcxproj]

    0 Warning(s)
    3 Error(s)

Time Elapsed 00:00:04.04
```

lovettchris, 2017-05-30:
The [Travis build passed ](https://travis-ci.org/Microsoft/AirSim/builds/237460722?utm_source=github_status&utm_medium=notification) after my commit, and the windows build.cmd passes on my machine, so your build error must be something else.  Perhaps you need to do a clean build or something because it looks like you are hitting a bug in the VC++ incremental compiler.  Notice how it says "Internal error" ? This usually means a compiler bug.

Mayankm96, 2017-05-31:
Yeah sorry about that. I missed that in the log. 

Well I am not sure but: 
```
==||=> squarebypath
==||=> Async RPC Error: rpclib: client error C0002: Function 'moveOnPath' was called with an invalid number of arguments. Expected: 7, got: 6
```

lovettchris, 2017-05-31:
I'm not seeing that, this could happen if DroneSHell.exe is out of date.  Are you using "build.cmd" ?  Perhaps you should do a "git clean -dfx" then run "build.cmd" and try again.

Mayankm96, 2017-05-31:
Sorry @lovettchris.. Something is still not right. 

```
==||=> circlebypath
==||=> Async RPC Error: rpclib: client error C0002: Function 'moveOnPath' was called with an invalid number of arguments. Expected: 7, got: 6

==||=> squarebypath
==||=> Async RPC Error: rpclib: client error C0002: Function 'moveOnPath' was called with an invalid number of arguments. Expected: 7, got: 6
```

lovettchris, 2017-05-31:
you probably also need to update your Unreal project with the latest AirSim bits, using something like ~\AirSim\Unreal\Environments\Blocks\update_from_git.bat.

Mayankm96, 2017-06-02:
The error still persists unfortunately. And I couldn't find the file you mentioned.

lovettchris, 2017-06-04:
Take a look at [clean_build.bat](https://github.com/Microsoft/AirSim/blob/master/Unreal/Environments/Blocks/clean_build.bat) and  
[update_from_git.bat](https://github.com/Microsoft/AirSim/blob/master/Unreal/Environments/Blocks/update_from_git.bat).  These show up to update your Unreal environment properly so that the new AirSim are installed properly.  This assume you have also used [build.bat](https://github.com/Microsoft/AirSim/blob/master/build.cmd) to build AirSim. this build.cmd doesn't just run msbuild, it also creates an xcopy deployable drop folder with all the bits in it that Unreal needs.

Mayankm96, 2017-06-10:
Yeah it is working now just fine. The order is important I think. 

Thanks again for your help @lovettchris !

closed 2017-06-10 by Mayankm96