microsoft/AirSim #246: Where can I find the camera parameters for the Left PIP Camera?

Mayankm96, 2017-05-30:


Dynode, 2017-05-30:
FOV is under BP_PIPCamera in UE4. Position & rotation (unless changed in code) is relative to the model in BP_FlyingPawn. The PIPCamera class holds some variables as well. The CameraDirector class regards setting views and rotation if you wish.

Mayankm96, 2017-05-31:
Hey @Dynode ! 

Thanksfor your help but I am not actually looking for position or orientation of the camera. Rather I need the intrinsic parameters of the camera which is being used so that I can create a point cloud from the images. 

I should have specified this earlier. Sorry for the confusion.

Dynode, 2017-05-31:
Then what you want is info about the projection method UE4 uses, the FOV and the viewport size, I think, as there is of course no actual lens or image chip. On the other hand, I have no experience with making point clouds from images. ;)

saihv, 2017-05-31:
I guess you can estimate the focal length through the relation

HFOV = 2 * arctan( width / 2f )

I assumed the principal point to be the center of the image, which will complete the intrinsics matrix.

Mayankm96, 2017-06-10:
I am closing this issue as you can see that the projection matrix for the stereo pair from [here](https://github.com/Microsoft/AirSim/blob/master/PythonClient/point_cloud.py)

```
projectionMatrix = np.array([[-0.501202762, 0.000000000, 0.000000000, 0.000000000],
                              [0.000000000, -0.501202762, 0.000000000, 0.000000000],
                              [0.000000000, 0.000000000, 10.00000000, 100.00000000],
                              [0.000000000, 0.000000000, -10.0000000, 0.000000000]])
```

closed 2017-06-10 by Mayankm96

ShukuiZhang, 2017-06-26:
Sorry to reopen this one @Mayankm96 . One thing I'm not quite clear is that in point_cloud.py, cv2.reprojectImageTo3D() takes the generated "depth" image as the input, while it is supposed to be "disparity" image according to the the opencv documentation [.](http://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html#reprojectimageto3d) So how do you actually use the "depth" image? Mimicing that python code gives me very weird results. Thanks in advance!

marcelinomalmeidan, 2017-07-06:
I am also having trouble reproducing the results from the [point_cloud.py](https://github.com/Microsoft/AirSim/blob/master/PythonClient/point_cloud.py). @lovettchris, could you let us know more about the conditions under which you got the projectionMatrix?
This matrix is supposed to change as resolution changes, right? What was the resolution in the disparity images?
Since we don't have access to the two camera images, we cannot reproduce the calibration step. Are there plans to allow us to have access to that? 
Did you perform intrinsic calibrations to obtain the camera parameters (fx, fy, cx, cy)?
What is the original baseline (distance between the cameras) in the depth sensor? I imagine this should be hardcoded somewhere...

Thanks!