microsoft/AirSim #247: Zooming or changing FOV for cameras

Dynode, 2017-05-30:
I know BP_PIPCamera in UE4 has a FOV parameter that affects all cams, but can we "zoom" just the main view or the PIPs? That would be nice. I've been unable to pinpoint relevant code for this.

lovettchris, 2017-06-02:
The field of view is configured using unreal editor, open the AirSim BluePrint PIP_Camera, open the camera component, and the details panel will show you the camera settings.  When we get the [UCameraComponent ](https://docs.unrealengine.com/latest/INT/API/Runtime/Engine/Camera/UCameraComponent/index.html) in code, you can also modify these settings at runtime, perhaps using AdditiveFOVOffset.

[image]

Dynode, 2017-06-02:
Alright, thanks. Fingers crossed for getting access to UCameraComponent in code, then.

sytelus, 2017-08-14:
We have added new settings for camera now including FOV: https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md#changing-resolution-and-camera-parameters

closed 2017-08-14 by sytelus