microsoft/AirSim #248: How to use Hello Drone

nickylimjj, 2017-05-31:
Hi,

How do I run the sample programs, such as Hello Drone?

I have tried opening `AirSim.sln` but I am not sure how to run the program.

Thank you!

Mayankm96, 2017-05-31:
Hey! 

You can run the executable file `HelloDrone.exe` directly. It is in the `AirSim/HelloDrone/build/x64/Debug` folder.

closed 2017-06-02 by lovettchris

rovdrcru, 2020-04-07:
Hello!

After executing the build.cmd I'm still unable to run the HelloDrone.exe. It seems to be waiting for a good connection.
I wasn't sure if this connection is internet connection since I'm connected with an ethernet-cable.

If not, what should it be connected to? 

[image]

"c:\Users\Robbe\Documents\AirSim\HelloDrone\build\x64\Debug>HelloDrone.exe
Waiting for connection - XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX..."

JBLip, 2021-12-09:
That doesn't seem to work for me. When I open the .exe file, the only thing I can see is: 
[image]

Can anyone please help me?