microsoft/AirSim #249: Adding a Surface Normal View to the type of views available

Mayankm96, 2017-05-31:
I saw in UnrealCV that it allows an additional view to have surface normals of the objects seen by the camera. As far as I know, AirSim uses the UnrealCV plugin to compute the depth map and segmented view. 

I am wondering how to incorporate the surface normal view into AirSim so that I can record that view from it along with the RGB view (which I suppose I can do from the APIs given). 

I would appreciate help on this matter. Thank you!

Mayankm96, 2017-05-31:
I tried installing the unrealCV plugin to my project. I couldn't see if it working or not because the rendering became really slow and it was really difficult to type on the UE terminal. 

Any ideas anyone?

Mayankm96, 2017-06-02:
So I tried making my own material for getting normals from the UE4Editor. The design is somewhat similar to what has been done in UnrealCV. However, I am now struggling to get the plugin working. I went through the `Source` folder of the AirSim plugin and added my own variables for rendering the view whereever I saw the other views being mentioned. But as expected I don't think it simply works like that. 

[image]

Mayankm96, 2017-06-10:
Well I did a cheap trick and it is working for me. If anyone ever needs to do this, feel free to comment on this issue.

closed 2017-06-10 by Mayankm96

weavermech, 2020-04-11:
> Well I did a cheap trick and it is working for me. If anyone ever needs to do this, feel free to comment on this issue.

I'm interested. I've made a material based on gradient but wonder if yours is better.