microsoft/AirSim #251: Bricks / Empty project crash on play button

isvogor-foi, 2017-06-01:
Hi,  I managed to compile everything on Ubuntu 16.04.2. UE4 works. When I try to run a Bricks example, or create an empty project and add AirSim plugin it crashes on pressing the play button. My stacktrace follows. I've been snooping around this issue section, but no solutions still. I'm not sure how to approach the issue. So, 1) I create an empty project, 2) I close it and add AirSim, 3) I change the dependencies in uproject file and start UE4 again, 4) it proceeds with compilation of 900 files, and after an hour or so of compiling, I press the play button and it crashes (I've set up the settings file, and it also doesn't connect to the pixhawk SITL). Any suggestions on where to start fixing this?
Thanks!
Ivan

Here is my stacktrace:

```
[2017.06.01-21.59.16:595][524]LogPlayLevel: PlayLevel: No blueprints needed recompiling
[2017.06.01-21.59.16:606][524]PIE: New page: PIE session: Untitled_1 (Jun 1, 2017, 1:59:16 PM)
[2017.06.01-21.59.16:606][524]LogPlayLevel: Creating play world package: /Temp/UEDPIE_0_Untitled_1
[2017.06.01-21.59.16:610][524]LogPlayLevel: PIE: StaticDuplicateObject took: (0.003724s)
[2017.06.01-21.59.16:610][524]LogAIModule: Creating AISystem for world Untitled_1
[2017.06.01-21.59.16:611][524]LogPlayLevel: PIE: World Init took: (0.001160s)
[2017.06.01-21.59.16:611][524]LogPlayLevel: PIE: Created PIE world by copying editor world from /Temp/Untitled_1.Untitled_1 to /Temp/UEDPIE_0_Untitled_1.Untitled_1 (0.005281s)
[2017.06.01-21.59.16:643][524]LogALAudio: ALAudio device opened : OpenAL Soft
[2017.06.01-21.59.16:645][524]LogALAudio: AL_VENDOR      : OpenAL Community
[2017.06.01-21.59.16:645][524]LogALAudio: AL_RENDERER    : OpenAL Soft
[2017.06.01-21.59.16:645][524]LogALAudio: AL_VERSION     : 1.1 ALSOFT 1.15.1
[2017.06.01-21.59.16:645][524]LogALAudio: AL_EXTENSIONS  : AL_EXT_ALAW AL_EXT_DOUBLE AL_EXT_EXPONENT_DISTANCE AL_EXT_FLOAT32 AL_EXT_IMA4 AL_EXT_LINEAR_DISTANCE AL_EXT_MCFORMATS AL_EXT_MULAW AL_EXT_MULAW_MCFORMATS AL_EXT_OFFSET AL_EXT_source_distance_model AL_LOKI_quadriphonic AL_SOFT_buffer_samples AL_SOFT_buffer_sub_data AL_SOFTX_deferred_updates AL_SOFT_direct_channels AL_SOFT_loop_points AL_SOFT_source_latency
[2017.06.01-21.59.16:645][524]LogInit: FAudioDevice initialized.
[2017.06.01-21.59.16:695][524]LogLoad: Game class is 'AirSimGameMode'
[2017.06.01-21.59.16:714][524]LogWorld: Bringing World /Temp/UEDPIE_0_Untitled_1.Untitled_1 up for play (max tick rate 0) at 2017.06.01-17.59.16
[2017.06.01-21.59.16:715][524]LogWorld: Bringing up level for play took: 0.001858
[2017.06.01-21.59.16:873][524]LogContentBrowser: Native class hierarchy updated for 'MovieSceneCapture' in 0.0012 seconds. Added 11 classes and 0 folders.
[2017.06.01-21.59.16:873][524]Cmd: r.CustomDepth 3
[2017.06.01-21.59.16:885][524]r.CustomDepth = "3"
Signal 11 caught.
Malloc Size=131076 LargeMemoryPoolOffset=131092 
CommonLinuxCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=196655 
[2017.06.01-21.59.17:113][524]LogCrashTracker: 

[2017.06.01-21.59.17:114][524]LogLinux: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x0000000000000002

[2017.06.01-21.59.17:114][524]LogLinux: Fatal error!

[Callstack]  00  0x00007f2b592e8f5f  FLinuxPlatformStackWalk::CaptureStackBackTrace(unsigned long long*, unsigned int, void*)
[Callstack]  01  0x00007f2b591d1f65  FGenericPlatformStackWalk::StackWalkAndDump(char*, unsigned long, int, void*)
[Callstack]  02  0x00007f2b592aa341  FLinuxCrashContext::CaptureStackTrace()
[Callstack]  03  0x00007f2b4f4ea1c0  CommonLinuxCrashHandler(FGenericCrashContext const&)
[Callstack]  04  0x00007f2b592ae4ba  PlatformCrashHandler(int, siginfo_t*, void*)
[Callstack]  05  0x00007f2b5efa5390  /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390) [0x7f2b5efa5390]
[Callstack]  06  0x00007f2b5924ed28  FMallocBinned::Private::FreeInternal(FMallocBinned&, void*)
[Callstack]  07  0x00007f2ae663f2da  std::__1::basic_filebuf<char, std::__1::char_traits<char> >::~basic_filebuf()
[Callstack]  08  0x00007f2ae6636ef9  msr::airlib::Settings::loadJSonFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >)
[Callstack]  09  0x00007f2ae6624ac1  ASimModeBase::initializeSettings()
[Callstack]  10  0x00007f2ae6624817  ASimModeBase::BeginPlay()
[Callstack]  11  0x00007f2ae662e442  ASimModeWorldMultiRotor::BeginPlay()
[Callstack]  12  0x00007f2b552a994c  AActor::DispatchBeginPlay()
[Callstack]  13  0x00007f2b552a8ec0  AActor::PostActorConstruction()
[Callstack]  14  0x00007f2b552a6d13  AActor::FinishSpawning(FTransform const&, bool, FComponentInstanceDataCache const*)
[Callstack]  15  0x00007f2b552a4cb5  AActor::PostSpawnInitialize(FTransform const&, AActor*, APawn*, bool, bool, bool)
[Callstack]  16  0x00007f2b55a901bd  UWorld::SpawnActor(UClass*, FTransform const*, FActorSpawnParameters const&)
[Callstack]  17  0x00007f2b55a908dc  UWorld::SpawnActor(UClass*, FVector const*, FRotator const*, FActorSpawnParameters const&)
[Callstack]  18  0x00007f2ae6623712  ASimHUD::BeginPlay()
[Callstack]  19  0x00007f2b552a994c  AActor::DispatchBeginPlay()
[Callstack]  20  0x00007f2b5635d947  AWorldSettings::NotifyBeginPlay()
[Callstack]  21  0x00007f2b55932dac  AGameStateBase::HandleBeginPlay()
[Callstack]  22  0x00007f2b5634b369  UWorld::BeginPlay()
[Callstack]  23  0x00007f2b558fcde9  UGameInstance::StartPlayInEditorGameInstance(ULocalPlayer*, FGameInstancePIEParameters const&)
[Callstack]  24  0x00007f2b50d58ed0  UEditorEngine::CreatePIEGameInstance(int, bool, bool, bool, bool, float)
[Callstack]  25  0x00007f2b50d5258a  UEditorEngine::PlayInEditor(UWorld*, bool)
[Callstack]  26  0x00007f2b50d467b0  UEditorEngine::StartQueuedPlayMapRequest()
[Callstack]  27  0x00007f2b50845411  UEditorEngine::Tick(float, bool)
[Callstack]  28  0x00007f2b51051a09  UUnrealEdEngine::Tick(float, bool)
[Callstack]  29  0x000000000041fffe  FEngineLoop::Tick() [Runtime/Core/Public/Stats/Stats2.h, line 1686]
[Callstack]  30  0x000000000042480a  GuardedMain(wchar_t const*) [/home/mistlab/Ivans/UnrealEngine-4.15.0-release/Engine/Source/Runtime/Launch/Private/Launch.cpp, line 166]
[Callstack]  31  0x00007f2b4f4eaf4b  CommonLinuxMain(int, char**, int (*)(wchar_t const*))
[Callstack]  32  0x00007f2b4e892830  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f2b4e892830]
[... 33 lines cut ...]
[2017.06.01-21.59.17:869][  0]LogInit:  - Memory allocator used: binned
[2017.06.01-21.59.17:869][  0]LogLinux: Skipped benchmarking clocks because the engine is running in a standalone program mode - CLOCK_REALTIME will be used.
[2017.06.01-21.59.17:869][  0]LogInit: Linux-specific commandline switches:
[2017.06.01-21.59.17:869][  0]LogInit:  -nodwarf (currently OFF): suppress parsing of DWARF debug info (callstacks will be generated faster, but won't have line numbers)
[2017.06.01-21.59.17:869][  0]LogInit:  -ansimalloc - use malloc()/free() from libc (useful for tools like valgrind and electric fence)
[2017.06.01-21.59.17:869][  0]LogInit:  -jemalloc - use jemalloc for all memory allocation
[2017.06.01-21.59.17:869][  0]LogInit:  -binnedmalloc - use binned malloc  for all memory allocation
[2017.06.01-21.59.17:869][  0]LogInit:  -httpproxy=ADDRESS:PORT - redirects HTTP requests to a proxy (only supported if compiled with libcurl)
[2017.06.01-21.59.17:869][  0]LogInit:  -reuseconn - allow libcurl to reuse HTTP connections (only matters if compiled with libcurl)
[2017.06.01-21.59.17:869][  0]LogInit:  -virtmemkb=NUMBER - sets process virtual memory (address space) limit (overrides VirtualMemoryLimitInKB value from .ini)
[2017.06.01-21.59.17:869][  0]LogInit:  - Physical RAM available (not considering process quota): 16 GB (16016 MB, 16400796 KB, 16794415104 bytes)
[2017.06.01-21.59.17:891][  0]LogUObjectArray: 146 objects as part of root set at end of initial load.
[2017.06.01-21.59.17:891][  0]LogUObjectAllocator: 24096 out of 0 bytes used by permanent object pool.
[2017.06.01-21.59.17:891][  0]LogUObjectArray: CloseDisregardForGC: 0/0 objects in disregard for GC pool
[2017.06.01-21.59.17:891][  0]LogTextLocalizationManager: No translations for ('') exist, falling back to 'en' for localization and internationalization data.
[2017.06.01-21.59.17:895][  0]CrashReportClientLog: CrashReportClientVersion=1.0
[2017.06.01-21.59.17:895][  0]CrashReportClientLog: CrashReportReceiver disabled
[2017.06.01-21.59.17:895][  0]CrashReportClientLog: DataRouterUrl: https://datarouter.ol.epicgames.com/datarouter/api/v1/public/data
[2017.06.01-21.59.17:895][  0]LogExit: Preparing to exit.
[2017.06.01-21.59.17:896][  0]LogObj: Freed 0b from 0 cluster array pools.
[2017.06.01-21.59.17:896][  0]LogExit: Object subsystem successfully closed.
[2017.06.01-21.59.17:937][  0]LogModuleManager: Shutting down and abandoning module CoreUObject (2)
[2017.06.01-21.59.17:937][  0]LogExit: Exiting.
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
```

andre-nguyen, 2017-06-01:
Pretty much a duplicate of #176 a mysterious crash where somehow a destructor is called when it shouldn't be or something like that. Personally I got it working by just commenting out json file handling and hard coding the settings but it looks like it doesn't work for everyone and sometimes it can crash again somewhere else. Also looks like some people just gave up when it segfaulted and then didn't post their stack trace so I'm unsure if my fix didn't work for them or if it just crashed somewhere else.

isvogor-foi, 2017-06-02:
I saw the thread, however since some people reported that the issue persists I didn't try it out. Apparently it worked once which is very weird. I'll try your suggestion and report back on it.

andre-nguyen, 2017-06-02:
It worked for me and one other guy, so your mileage may vary but for Alex it crashed at some other place. And for what its worth, it worked on two different computers on my side.

My guess is that there are multiple issues but some people sort of bunched up all the segfaults into one issue and said it doesn't work on Ubuntu. Post your new stack trace if it crashes again.

isvogor-foi, 2017-06-02:
Will do! - also, I just realized we work in the same building which is good for me if I don't get it to work :rofl:

isvogor-foi, 2017-06-02:
Hm... (trying on two different PCs), it seems that it worked once, and now it crashes again. Hmmmmm.... 
`[2017.06.02-15.27.19:469][218]LogLinux: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x0000000000000000

[2017.06.02-15.27.19:469][218]LogLinux: Fatal error!

[Callstack]  00  0x00007f02c945c8bf  FLinuxPlatformStackWalk::CaptureStackBackTrace(unsigned long long*, unsigned int, void*)
[Callstack]  01  0x00007f02c9316c95  FGenericPlatformStackWalk::StackWalkAndDump(char*, unsigned long, int, void*)
[Callstack]  02  0x00007f02c9414671  FLinuxCrashContext::CaptureStackTrace()
[Callstack]  03  0x00007f02be461340  CommonLinuxCrashHandler(FGenericCrashContext const&)
[Callstack]  04  0x00007f02c9419572  PlatformCrashHandler(int, siginfo_t*, void*)
[Callstack]  05  0x00007f02cf240390  /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390) [0x7f02cf240390]
[Callstack]  06  0x00007f02bd85f6ef  /lib/x86_64-linux-gnu/libc.so.6(+0xa56ef) [0x7f02bd85f6ef]
[Callstack]  07  0x00007f025945036f  getWifiRssi(int, char const*)
[Callstack]  08  0x00007f025944fbb0  UdpClientPort::UdpSocketImpl::getRssi(char const*)
[Callstack]  09  0x00007f025944f21c  UdpClientPort::getRssi(char const*)
[Callstack]  10  0x00007f0259408f90  mavlinkcom_impl::MavLinkConnectionImpl::getTelemetry(mavlinkcom::MavLinkTelemetry&)
[Callstack]  11  0x00007f025939acdc  mavlinkcom::MavLinkConnection::getTelemetry(mavlinkcom::MavLinkTelemetry&)
[Callstack]  12  0x00007f025928e018  msr::airlib::MavLinkDroneController::impl::reportTelemetry(float)
[Callstack]  13  0x00007f0259289ffe  msr::airlib::MavLinkDroneController::reportTelemetry(float)
[Callstack]  14  0x00007f0259209292  MultiRotorConnector::updateRendering(float)
[Callstack]  15  0x00007f0259211b6f  ASimModeWorldBase::Tick(float)
[Callstack]  16  0x00007f025921a3fc  ASimModeWorldMultiRotor::Tick(float)
[Callstack]  17  0x00007f02c4c1642f  FActorTickFunction::ExecuteTick(float, ELevelTick, ENamedThreads::Type, TRefCountPtr<FGraphEvent> const&)
[Callstack]  18  0x00007f02c5d22bab  FTickFunctionTask::DoTask(ENamedThreads::Type, TRefCountPtr<FGraphEvent> const&)
[Callstack]  19  0x00007f02c5d22507  TGraphTask<FTickFunctionTask>::ExecuteTask(TArray<FBaseGraphTask*, FDefaultAllocator>&, ENamedThreads::Type)
[Callstack]  20  0x00007f02c933777d  FNamedTaskThread::ProcessTasksNamedThread(int, bool)
[Callstack]  21  0x00007f02c93359d6  FNamedTaskThread::ProcessTasksUntilQuit(int)
[Callstack]  22  0x00007f02c93337ab  FTaskGraphImplementation::WaitUntilTasksComplete(TArray<TRefCountPtr<FGraphEvent>, TInlineAllocator<4u, FDefaultAllocator> > const&, ENamedThreads::Type)
[Callstack]  23  0x00007f02c5d1cf03  FTickTaskSequencer::ReleaseTickGroup(ETickingGroup, bool)
[Callstack]  24  0x00007f02c5d1645e  FTickTaskManager::RunTickGroup(ETickingGroup, bool)
[Callstack]  25  0x00007f02c551fe88  UWorld::RunTickGroup(ETickingGroup, bool)
[Callstack]  26  0x00007f02c5526fdf  UWorld::Tick(ELevelTick, float)
[Callstack]  27  0x00007f02bf8df509  UEditorEngine::Tick(float, bool)
[Callstack]  28  0x00007f02c02b0769  UUnrealEdEngine::Tick(float, bool)
[Callstack]  29  0x0000000000423de0  FEngineLoop::Tick() [Runtime/Core/Public/Stats/Stats2.h, line 1686]
[Callstack]  30  0x000000000042b923  GuardedMain(wchar_t const*) [/home/mistlab/Ivans/UnrealEngine-4.15.0-release/Engine/Source/Runtime/Launch/Private/Launch.cpp, line 62]
[Callstack]  31  0x00007f02be4620ca  CommonLinuxMain(int, char**, int (*)(wchar_t const*))
[Callstack]  32  0x00007f02bd7da830  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f02bd7da830]
[Callstack]  33  0x0000000000415e29  /home/mistlab/Ivans/UnrealEngine-4.15.0-release/Engine/Binaries/Linux/UE4Editor(_start+0x29) [0x415e29]

`

andre-nguyen, 2017-06-02:
Right so you're at the same crash Alex got to.

My guess is that this crash has to do with an uninitialized pointer.

Si this patch line https://github.com/kersny/airsim_nvidia_docker/blob/28256e03af56d800a37f19b93f40fa7b5983bee0/airsim2.patch#L35

@kersny initializes `wifiInterfaceName` to `nullptr`, 
```patch
 	class MavLinkTelemetry : public MavLinkMessageBase {
 	public:
 		const static uint8_t kMessageId = 204; // in the user range 180-229.
-		MavLinkTelemetry() { msgid = kMessageId; }
+		MavLinkTelemetry() : wifiInterfaceName(nullptr) { msgid = kMessageId; }
 		uint32_t messagesSent;		 // number of messages sent since the last telemetry message
 		uint32_t messagesReceived;	 // number of messages received since the last telemetry message
uint32_t messagesHandled; // number of messages handled since the last telemetry message
```

This should catch your crash here https://github.com/Microsoft/AirSim/blob/a1d4a62f403f1e29a3b43435c65a44392ef89c99/MavLinkCom/src/impl/MavLinkConnectionImpl.cpp#L532 with the `nullptr` check
```c++
    if (telemetry_.wifiInterfaceName != nullptr) {
        telemetry_.wifiRssi = port->getRssi(telemetry_.wifiInterfaceName);
}
```

You'll notice that `
void MavLinkConnectionImpl::getTelemetry(MavLinkTelemetry& result)` is the 10th frame in your call stack.

lovettchris, 2017-06-02:
thanks for the fix, that could explain a lot.  I'll get this pushed into master.

isvogor-foi, 2017-06-05:
@andre-nguyen your fix with files seems to work. I've managed to takeoff using px4 console. However, on one of my two computers, I can only run it once. I'm investigating this issue and I'll get back here once I figure exact nature of the crash.

isvogor-foi, 2017-06-05:
It seems that applying Andre's fix for opening JSON file, along with all patches here https://github.com/kersny/airsim_nvidia_docker/tree/28256e03af56d800a37f19b93f40fa7b5983bee0 makes it work. 
However, there is no way to exit without a crash. It usually end's up by me killing it from console.

andre-nguyen, 2017-06-05:
Do you get a stack trace on the exit crash?

isvogor-foi, 2017-06-06:
Yup...
Looks like UDP issue, as if I use HITL and not SITL?

```
LinkConnectionImpl: Error handling message 0 on connection 'hil', details: MavLinkConnectionImpl: Error sending message on connection 'Proxy to: LogViewerOut at 127.0.0.1:14389', details: UdpClientPort socket send failed with error: 111

MavLinkConnectionImpl: Error handling message 147 on connection 'sitl', details: MavLinkConnectionImpl: Error sending message on connection 'Proxy to: LogViewer at 127.0.0.1:14388', details: UdpClientPort socket send failed with error: 111

MavLinkConnectionImpl: Error handling message 147 on connection 'sitl', details: MavLinkConnectionImpl: Error sending message on connection 'Proxy to: QGC at 127.0.0.1:14550', details: UdpClientPort socket send failed with error: 111

*** Error in `/home/mistlab/Ivans/UnrealEngine-4.15.0-release/Engine/Binaries/Linux/UE4Editor': free(): invalid pointer: 0x00007fac2bbef930 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7faca02a97e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x7fe0a)[0x7faca02b1e0a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7faca02b598c]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x6439)[0x7facb1cad439]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x7870)[0x7facb1cae870]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7faca033882d]

```

andre-nguyen, 2017-06-06:
I would be more curious to know why anything would be sent on exit to the LogViewer and to QGC. So connection error 111 means connection refused. Why is the connection being opened when we are closing the application?

@lovettchris I followed the code for the `MavLinkConnection` destruction and I can't see why it would suddenly throw here https://github.com/Microsoft/AirSim/blob/a1d4a62f403f1e29a3b43435c65a44392ef89c99/MavLinkCom/src/impl/MavLinkConnectionImpl.cpp#L492 but only on exit and not during operation. Any ideas?

clovett, 2017-06-06:
that might be a normal shutdown condition, hence the "catch" block.

andre-nguyen, 2017-06-08:
Soooo expected behaviour, @isvogor-foi maybe you can close this issue then :laughing:

closed 2017-06-10 by lovettchris