microsoft/AirSim #252: MavLinkCom does not work on OS X

nacho-carnicero, 2017-06-02:
Hi, 

I've been trying to use the MavLinkCom library on my Mac without any success. I wanted to test the MavLinkTest main program and check if it worked. I compiled it by performing `cmake MavLinkCom` and then `make` and then ran the executable that was created in `cmake/output/bin`. Nevertheless when I try to connect to a simulation that is running (PX4 SITL and jMAVSim) through a UDP client socket with flag `-udp:127.0.0.1:14550` I get a socket error as follows:
```bash
./MavLinkTest -udp:127.0.0.1:14550
Connecting to offboard drone at address 127.0.0.1:14550
libc++abi.dylib: terminating with uncaught exception of type std::runtime_error: MavLinkConnectionImpl: Error sending message on connection 'drone', details: UdpClientPort socket send failed with error: 56
```

Error code 56 is issued when the socket is already connected and you try to connect it again (you can check it [here](https://www.ibm.com/support/knowledgecenter/SSLTBW_1.13.0/com.ibm.zos.r13.cs3cod0/syserret.htm)).

This does not happen if I connect through a `-server` connection.

Besides, I've tried to compile and execute this on both Windows and Linux, and it works just fine with both client and server connections. I guess then it's an OS X specific issue. I've been working on a fix, I have to test it and I will issue a Pull Request for this

lovettchris, 2017-06-02:
Is another app already using port 14550?  For example, QGroundControl ?

nacho-carnicero, 2017-06-02:
No, no app is using it. I've issued Pull Request #253 , I really think it's something specific to OSX and the way it handles UDP client sockets.

nacho-carnicero, 2017-06-02:
Also I was thinking, since MavLinkCom can be used on its own for other projects, maybe it would be useful to put all the MavLinkCom cmake files in that folder so that it is more self-contained?

lovettchris, 2017-06-02:
The only reason I didn't do that is so folks didn't accidentally clobber the hand crafted visual studio *.vcxproj files by running cmake on windows.  But I don't feel too strongly about it.  

Can you run MavLinkTest app from terminal and talk to PX4 using -serial:/dev/ttyACM0 ?  Then can you run MavLinkTest with  -serial:/dev/... and -proxy 127.0.0.1:14550 and connect to QGroundControl without AirSim ?

lovettchris, 2017-06-02:
Oh, saw your pull request, looks like you fixed it then? Thanks.

nacho-carnicero, 2017-06-02:
Yes with the commit of the PR I'm able to run MavLinkTest without errors.

Thanks for the merge, I'm closing this issue.

closed 2017-06-02 by nacho-carnicero