microsoft/AirSim #254: Best way to fly AirSim drone from ground pilot perspective

Timebutt, 2017-06-02:
I have been playing around with a solution to be able to fly the drone in AirSim from a pilot's perspective. This can be interesting for many other people, as AirSim could excellently serve as a flight trainer as well. I have my official UAV pilot examination coming up shortly so could use some extra flight time ;)

I present my current solution to the problem to start a discussion, and see if there maybe is a more elegant and user-friendly solution. Maybe a custom AirSim component or specific behaviour could natively provide this functionality instead of hacking around like I do now?

Here goes:
I first made sure there is no `Player Start` in the Unreal environment, and add the AirSim pawn. Next, I add a traditional camera (so not the `BP_PIPCamera`)

[image]

Next I open the Level Blueprint:

[image]

There I added the following modules and connections to make sure that the new camera always center focuses on our drone, and that the newly added camera is used as the default camera. The white line arriving comes from `Event Tick` module, so that this code is continuously executed. A solution where I execute ´Set View Target with Blend` once (`Event BeginPlay`or `Respawn player_Event` for instance) didn't seem to work correctly. Having this code run every tick probably isn't the most efficient solution, but it works.

[image]

Result: flying as if you were flying the drone outside in real life!
(in the image the 'pilot' is higher up than the drone)

[image]

Is this the way to go, or can we improve this?

lovettchris, 2017-06-02:
This looks great.  We have these modes in the ~\AirSim\Unreal\Plugins\AirSim\Source\CameraDirector.cpp.  We also have this already: ACameraDirector::inputEventGroundView which is bound to the backslash key:
`
    UAirBlueprintLib::BindActionToKey("inputEventGroundView", EKeys::Backslash, this, &ACameraDirector::inputEventGroundView);
`
Perhaps that info is missing from the F1 help...

Timebutt, 2017-06-06:
Thanks for the hint and reference to the code. You are right though: that specific info is available in the F1 menu, I must have somehow looked over it.
Still, the `Ground Observer View` poses a problem, as illustrates following screenshot:

[image]

Apparently, the `Ground Observer View` sets `z = 0`. The spot I chose to take off here in _Landscape Mountains_ is on a road somewhere in the middle of the map (`z = 12354.526367 cm`). We can clearly see the entire landscape disappeared because we are below it, and the drone is very far away. The view isn't exactly very usable in this way.

I guess an optimal solution would set the observer `z` equal to the AirSim drone takeoff  `z`, probably solving the issue at hand. This would probably have weird consequences for the camera view at takeoff, so placing the 'observer' a little bit next to the drone could maybe be a bit better. I'll experiment a bit and report back my findings.

Timebutt, 2017-06-06:
Another problem with the `Ground Observer View´, is the fact that it follows the drone and stays below it. A 'real life' UAV doesn't often move around, and rarely has the speed to follow a drone at high speed ;)

In any case, I submitted a pull request where I present the 'Pilot' view functionality, this time neatly implemented in C++ as opposed to messing around with cameras as I first described. I mapped the functionality to the '/' key and updated the HUD widget to reflect the new functionality. 

Idea is, that the starting location of the `CameraDirector` is the 'head' of the person flying the drone. This prevents issues as described in my last post, and allows the user to efficiently place the 'pilot' in the scenario. Come to think of it: I haven't tested what happens if no director is placed and a user uses the `PlayerStart` object. Will test this first thing.

closed 2020-08-28 by jonyMarino