microsoft/AirSim #256: Error in python API to takeoff

Mayankm96, 2017-06-02:
I am running the python API script for takeoff (`takeoff.py`) and I am getting the following error: 
```
Waiting for home GPS location to be set...
Home lat=47.6415, lon=-122.14, alt=121.326
Traceback (most recent call last):
  File "takeoff.py", line 22, in <module>
    if (client.getLandedState() == LandedState.Landed):
  File "C:\Users\xam\Desktop\AirSim\PythonClient\PythonClient.py", line 61, in getLandedState
    return self.client.call('getLandedState')
  File "C:\Python27\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\Python27\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: server could not find function 'getLandedState' with argument count 0.
```

Is anyone else also having the same problem?

lovettchris, 2017-06-02:
This would mean you need to rebuild, some boys are your of date.  I use build.cmd to ensure correct bits, followed by an xcooy to my unreal environment, then a clean rebuild of my unreal project.   see the unreal/environment/blocks/update_from_git.bat.  You should do this every time you update airsim code.

Mayankm96, 2017-06-03:
Seems like it's working now. I'll keep the rebuilding tip in mind.

closed 2017-06-03 by Mayankm96

yongwon2, 2017-08-10:
I don't know how to build to build.cmd