microsoft/AirSim #257: moveByVelocity in Python doesn't work properly

RodgerLuo, 2017-06-02:
I want to use moveByVelocity function in Python to move the drone vertically. I've tried command "moveByVelocity -vz -1" in DroneShell, and it works fine by slowing raising up the drone. However, when I tried "client.moveByVelocity(0, 0, 1, 5, DrivetrainType.MaxDegreeOfFreedom, YawMode(False, 0))", the drone didn't move at all. Other functions, like takeoff, or moveToZ work fine in Python. Did I set parameters wrong in client.moveByVelocity()? 

Thank you!

lovettchris, 2017-06-04:
I just tried it and it worked fine.  Note that a 'vz' of 1 means fly downwards since AirSim uses the NED coordinate system.  If you meant to fly upwards use vz=-1.

RodgerLuo, 2017-06-06:
Thank you! It turns out that I put client.hover() right after client.moveByVelocity() to cause the issue.

closed 2017-06-06 by RodgerLuo