microsoft/AirSim #258: Error when launching the project.

doriand1122, 2017-06-05:
Hello,
I'm following this video tutorial about AirSim: https://www.youtube.com/watch?v=1oY8Qu5maQQ
Hovewer I get an error when launching the project: The following modules are missing or built with a different engine version: UE4Editor-AirSim.dll. Would you like to rebuild them now?
If I choose Yes, there's another error: MyProject could not be compiled. Try rebuilding from source manually.
If I choose No, nothing happens.
Can anyone help me and point what can be wrong and how to solve this issue? My Unreal version is 4.15.2. I'm using AirSim from repo: https://github.com/Microsoft/AirSim
Thanks.

lukaszdworako, 2017-06-05:
Can you tell me what files are present in the Plugins/Airsim folder that you put in the project file?

ethanyqc, 2017-06-06:
I encounter the same problem, the files are Content, Dependencies, Source and AirSim.uplugin

doriand1122, 2017-06-06:
I present screens.
[image]
[image]
[image]
[image]

wang19940601, 2017-06-07:
I encounter the same problem！

lukaszdworako, 2017-06-07:
I see we are getting the same issue for different reasons then, I receive this issue since the build was incomplete/ not all files were present.

Mayankm96, 2017-06-07:
My project is building when I launch it through the UE4Editor. However, when I run the simulation, the application is crashing.

Mayankm96, 2017-06-07:
I tried doing everything from the start, but now I get a new error on starting the project:

[image]

lukaszdworako, 2017-06-07:
This is odd, just pulled latest build and tried again and it worked. Can't give any advice since this seems to have just been some sort of anomaly.

lovettchris, 2017-06-07:
perhaps it was GitHub, they were down yesterday... which was rather shocking.

Mayankm96, 2017-06-07:
I am still receiving the second error I shared regarding the COOKED and UNCOOKED version. The following is what I did:
1. Delete the existing repository
2. Clones the git repo again
3. Build the AirSim plugin using VS Commmand Prompt (Build successful)
4. Copied the folder to my project
5. Open the project `.sln` file on VS2015 and ran it. 
 
These steps always worked before. I am not sure where I am going wrong now.

Mayankm96, 2017-06-08:
Oh well I suppose the correct steps to build the plugin is:
1. Clone the git repo
2. Run `clean.cmd`
3. Run the two scripts in `Unreal/Environment/Blocks` i.e. `clean_build.cmd` and `update_from_build.cmd`
4. Run the `build.cmd`
5. Copy the plugin folder to your project

The above worked for me at least. I hope this is the suggested way as well. @lovettchris

wang19940601, 2017-06-08:
Hello, somebody can send me a copy of the Plugins/Airsim. my email address is wangshubo@cau.edu.cn, thank you very much

lovettchris, 2017-06-08:
after running build.cmd, ~\unreal\plugins\airsim will be populated.

closed 2017-06-10 by lovettchris

QiuMingZS, 2020-02-15:
I also met this problem using Unreal Engine 4.24.0, so I changed it to 4.18.1, the problem was then solved.