microsoft/AirSim #259: Error with build.cmd

lukaszdworako, 2017-06-05:
It seems as though the current version of AirSim has some kind of error in the build.cmd as it is not generating all AirSim plugin files. The only file my build.cmd is producing is Unreal/Plugins/AirSim/Source, thus leading to errors since not all required files are required to generate the visual studio project properly. This seems to result in errors such as issue#258[https://github.com/Microsoft/AirSim/issues/258](url)

Timebutt, 2017-06-06:
I just pulled the latest version of AirSim and built it using `build.cmd` with no issue. Are you using the 
`Developer Command Prompt for VS2015`, as is prescribed?

[image]

lukaszdworako, 2017-06-07:
I am using the Developer Command prompt and have this issue. I have made modifications to the code however they should not impact the make since there are no changes to file names or locations. I will try pulling the latest build and see if I still have the same problem with my code.

lovettchris, 2017-06-07:
If your issue is the same as [#258 ](https://github.com/Microsoft/AirSim/issues/258) can we close this one and just use that thread?

lukaszdworako, 2017-06-07:
Seemed to have some sort of anomaly, working now

closed 2017-06-07 by lukaszdworako